CTCaer/hekate #126: Black screen after the splash screen

dothackjhe, 2018-10-03:
As of version 4.2, I'm getting a "black screen" issue booting the bootloader after the splash screen.

However, the issue is circumvented by pressing the "power button" which essentially terminates the initial connection made with the Switch via TegraSmashGUI. Re-injecting the Hekate 4.2 payload at this point fixes it. I'm not sure if this is an issue with the configuration, but I'm leaning towards it.

More importantly, is there any way around this?

CTCaer, 2018-10-03:
> This is 
Auto full power off when the device woke up from HOS' power off
Usefull with modchips/dongles when using AutoRCM. (You can see it as a breathing backlight with hekate's logo).
>
>It basically detects that the console woke up from an alarm set by HOS and powers off to protect your battery and not stay in Autorcm (as long as you inject the payload ofc).
>
>In 4.2, there's an option to disable it, if it does not meet your use case. Options -> Auto HOS power off -> Disable.

dothackjhe, 2018-10-03:
What do you mean by "HOS?" What is it, basically?

noirscape, 2018-10-03:
@dothackjhe - HOS = Horizon OS. It's the Switch's operating system.