CTCaer/hekate #128: SD Card and eMMC data (@lba 01152000), do not match

sanglongbest, 2018-10-11:
When I try to perform a full backup of my nand on My SanDisk Ultra 128GB, always show a error message:

SD Card and eMMC data (@lba 01152000), do not match!
Verification failed..

Press any key and try again...

I have tried to use the hekate 3.2, 4.1 and 4.2 on the Fat32 and exFat. Could you help me to find the problem, please?

CTCaer, 2018-10-11:
Sorry to inform you, but your sd card is 99% fake.

You can see this from non-proper codes in Console info -> Sd card info
and by testing with h2testw or other tool like this.

sanglongbest, 2018-10-11:
Thank you very much. I will test as soon as possible. Sometime, local shopper is not believeable.

sanglongbest, 2018-10-11:
Could I ask which SD card is better, sansumg or SanDisk? I would buy a new one on Amazon now.

ErichMoraga, 2018-10-11:
Amazon is riddled with fakes, which I've discovered the hard way.  Retail store or straight from mfg. is best.

sanglongbest, 2018-10-11:
> Amazon is riddled with fakes, which I've discovered the hard way. Retail store or straight from mfg. is best.

I think you are right. I need to buy new from the official website.

closed 2018-11-12 by CTCaer

lisenxd, 2019-03-10:
Has the problem been solved after you change a new SD card?

eALduderino, 2019-03-21:
Hey guys i have the same problem but with (@LBA 00800000)
Pretty sure my SD Card is 100% official SanDisk extreme 128GB SDXC Card since this worked before.
I recently reset my switch and wanted to dumpe the factory restored nand when this error occured.
Any advise?

eALduderino, 2019-03-21:
for the record i sued hekate 4.9.1 on FW 5.1.0

eALduderino, 2019-03-21:
I also formated the sd card with the sd card formatter software, still didnt work.

CTCaer, 2019-03-21:
You sure you used 4.9.1?

eALduderino, 2019-03-21:
yes im sure, i just downloaded it from your github. i am back to 4.6 at it seems it working out since i'm pass the 16% mark it it errored out twice before. i hope it will run until it finishes. Still, if you have any advice i'll be gladf to test it with 4.9.1 again...
PS thx for the quick reply dude

eALduderino, 2019-03-21:
to be fully transparent, though. I discovered i lost my clean nand dump. then i followed  Nevercholt Tech's instructions on YouTube to restore the switch to factory without a clean nand dump. before that i used my hekate 4.6 to do a dirty nand dump since it was already on my sd card. everything worked out fine. then afetr restoring the switch i tried to dump the restored nand with 4.9.1 again which gave me this error. could it be an issue with the FW 5.1.0?

CTCaer, 2019-03-21:
It doesn't matter what you backup as long as backed up data in sd match the re reads of emmc.
Check if you see hekate v4.9.1 in main menu.

Also in your case I would do a hw2testw so you know it's legit. Fake cards have real storage but it's smaller than real. So it may not happen immediately.

Additionally please take a photo via `Console info -> SD card info` and up it here.

CTCaer, 2019-03-21:
@eALduderino Additionally can you tell me, what's your Verification option (Sparse or Full? You can see it in Options)

eALduderino, 2019-03-21:
dump just finished with 4.6 i will download 4.9.1 again and try it immediately as soon as i copied the dump off my sd card. after that ill send you a pic of the sd card info and will also tst the card with h2testw_1.4

CTCaer, 2019-03-21:
Don't try to dump now that you got your dump.

Just get a pic of sd card info and check tools -> verification options and tell me which one has the *asterisk

eALduderino, 2019-03-21:
So i'm afraid i seriously effed sth up, the dump worked this time after redownloading 4.9.1 again. Sorry fur making such a fuzz since i obviously messed it up somehow. Anyways here you get the image of the sd info. maybe the file got corrupted during downloading or unpacking or i simply used the wrong version, although i replaced the old version with the new one.  I am also running the h2test right now. i guess it was just user error :(
[image]

CTCaer, 2019-03-21:
It's ok. It's important for that feature to work.

You can stop h2testw.
Your sd card is legit from the info I'm seeing. (And a beefy one also)

eALduderino, 2019-03-21:
Yes exactly, thats my card. I let the test run just to be sure, no errors found. Thanks again m8 for the grat support. And especially thank you for putting all the effort into the community, we appreciate it! XOXO :)

fennectech, 2019-03-21:
You have synthisized a nand image correct?   You should be able to write it to the nand with memloader using a program like etcher or dd

eALduderino, 2019-03-21:
i dont quite get the question. I didn't want to restore ar write a nand dump to my switch, i wanted to create a dump and failed due to using the wrong hekate version probably. What do you mean by synthisized exactly?

fennectech, 2019-03-21:
Oh.  I thought you were trying to restore a brick.    What’s going on?

eALduderino, 2019-03-21:
no it wasnt bricked, i just lost my clean nand dump. after cleaning the nand and factory-restoring the system without a clean nand dump, i now successfully dumped the functioning clean nand. So, Like Ice Cube would say..  "Today was a good Day!" Thanks for caring m8!

fennectech, 2019-03-21:
Glad to hear it.    I had a fatal happen while downgrading with choidujorNX.   So.   I had to restore a nand backup.   About shit myself when my switch refused to boot tho.