CTCaer/hekate #129: [Enhancement] Add option to auto

Rickardo987, 2018-10-15:


lunalik2, 2018-10-18:
I'll try to implement this, then make PR. No promises though.

matt-teix, 2018-12-05:
You can add the payload to the .ini and autoboot that way

Rickardo987, 2018-12-05:
...Did you read the post?

XiaoNiaoZZJ, 2020-05-07:
add
[EMUORG]
payload=bootloader/payloads/fusee-primary.bin
to your hekate_ipl.ini and set autoboot=<1based-index to this "EMUORG">
It works well.

its my hekate_ipl.ini
[config]
autoboot=1
autoboot_list=0
bootwait=3
backlight=100
autohosoff=0
autonogc=1
updater2p=1

[EMUORG]
payload=bootloader/payloads/fusee-primary.bin

[EMUDEF]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
emummcforce=1
atmosphere=1
icon=bootloader/res/icon_payload.bmp
logopath=bootloader/bootlogo.bmp

[STOCK]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1

masagrator, 2020-05-07:
Helping with 1.5 year old issue. Nice. :P

CTCaer, 2022-05-17:
I wonder why I never closed that.. hmm

closed 2022-05-17 by CTCaer