CTCaer/hekate #134: Can't read/write from/to SD card

rapsaGnauJ, 2018-10-22:
Hekate doesn't backup anything into my SD card. Sometimes it "finishes succesfully" but then there isn't even the backup folder, sometimes it gives me "Error (5) creating file" and sometimes it stops mid-backup. Also can't open /bootloader/hekate_ipl.ini to launch into CFW. The SD card is 128 GB if that matters.

CTCaer, 2018-10-22:
It seems like your card might be fake.

Can you post a photo of Console info -> Sd card info?

And even better test it with an app like h2testw (after formatting it)

rapsaGnauJ, 2018-10-22:
Here is the requested picture. I'm passing the test right now, I'll share results in an hour or so. However, I don't think my card is fake because I nearly filled it up sometime. It also works perfectly on PC and Switch. 
[image]

CTCaer, 2018-10-22:
Fake always work on writing/reading. But when they fill they just handle garbage data on reads.

It's seems like a genuine Sandisk.
Although, I didn't knew that they still made Class 10 with no U1 or U3.
That's strange.

Normally since many years now, the cards are at least U1. (Don't confuse it with UHS-I. UHS-I is mandatory for U1/U3 rating.)

rapsaGnauJ, 2018-10-23:
Trying a 4GB class 4 micro SD card. Seems to work ok. The problem must be the card somehow.

closed 2018-11-08 by CTCaer