CTCaer/hekate #139: "Failed to init SD Card. Make sure it is inserted"

junyjeffers, 2018-11-04:
I am trying to run Lakka on my switch, everything else is working but when I try to click Launch, nothing happens, I have tried to run Lakka on both 4.2 and 4.0.  My switch Detects my SD Card, I can use it for games and stuff, but not hekate. Any fix? I am currently using 4.0 on 6.0.1 firmware

junyjeffers, 2018-11-04:
I also have a 32GB Samsung EVO Select Micro SDHC Card, it works on my cpu, but my switch does not even detect it... and help?

CTCaer, 2018-11-04:
This is because the sd card reader is not seated well or misses a pin.

Hekate does not support 1bit data bus (which is where the card defaults if it can't use 4bit data bus).
Some cards have problems with 1bit data bus, that's why your samsung evo is not recognized.

You need to check the connector and maybe add more flexible padding on top of it.

CTCaer, 2018-11-08:
To not reply, I suppose you fixed your sd card reader.

closed 2018-11-08 by CTCaer