CTCaer/hekate #143: Printing TSEC keys throws error

agentcatcat, 2018-11-11:
trying to print the TSEC keys gives me this error:
```
Could not identify package1 version 
to read TSEC firmware <= '20180802162753'>.
```
any idea what is happening here, or how to fix it?

CTCaer, 2018-11-11:
hekate version?

agentcatcat, 2018-11-11:
4.2

CTCaer, 2018-11-11:
hmm looks like your BOOT0 is corrupt.
You can boot in HOS?

agentcatcat, 2018-11-11:
yes

EternityShack, 2018-11-12:
What do you mean you think you can?

agentcatcat, 2018-11-12:
i hadnt checked properly after i got the error, now i have
yes i can boot into HOS

CTCaer, 2018-11-12:
That's the 6.0.0-6.1.0 pkg1.1 version.

Which is already supported since v4.1.
So dunno where's the problem..

(it works on 6.x.x)

Endar-Marit, 2018-11-20:
I'm getting a similar error with 6.2 when trying to dump the tsec keys 

 `Unknown package1 version for reading TSEC firmware (= '20181100705733'). `

Everything else can be dumped bar the tsec keys and package 1/2

CTCaer, 2018-11-20:
6.2 is not supported yet.
The crypto moved to tsec fw which is encrypted with keys residing in TSEC co-processor.

The new tsec fw needs a specific sanitized environment to work. Basically it checks everything.

closed 2018-11-20 by CTCaer

CanadianHummus, 2019-04-24:
I am on 8.0.0 using Hekate V2.5 and when I try to print my keys I get 

"Found pkg1 <'################'>

Uknown pkg1 version for reading TSEC firmware"

ELY3M, 2019-04-24:
You need to update your hekate to 4.10.1

flippnswithez, 2019-06-20:
i am on 8.1 using hekate 4.10.1 and get the same error