CTCaer/hekate #147: 6.2.0

melsbacksfriend, 2018-11-21:
Are you aware that 6.2 breaks this?  I want you to try your best to add 6.2 support.  I was foolish enough to update.  I was like "Oh, it's just another update."

CTCaer, 2018-11-21:
[image]

closed 2018-11-21 by CTCaer

melsbacksfriend, 2018-11-21:
Not to mention, the update also blows a fuse so no downgrading.  :(

fennectech, 2018-11-21:
We can always downgrade.   (Unless the cart slot gets a fuse blown.

melsbacksfriend, 2018-11-22:
I literally said that the update blows a fuse.  I tried downgrading and now I need to replace the cpu to unbrick my switch :(

fennectech, 2018-11-22:
You have to use hekate to boot your previous firmware. That will bypass the
fuse checks. (Other than the cart reader one)

melsbacksfriend, 2018-11-23:
@fennectech Thank you, I didn't need to get better at soldering because of you.

fennectech, 2018-11-23:
@melsbacksfriend  IF 6.2 burned a fuse in the cart reader your cart slot wont work until you update.   There’s no workaround at current.   And.  Replacing the SOC would do nothing for your switch sadly.   You would have to find a way to rebuild the NAND image and keys from scratch.  (Wich without nintendo’s Help initializing the unit your not doing)