CTCaer/hekate #148: Print TSEC keys error after rawnand.bin restore from 6.2 to 6.1

pjh4, 2018-11-24:
Hello,

I accidentally updated to 6.2 so I tried downgrading back to 6.1 with my rawnand.bin file. Unfortunately I didn't save the boot0/1 file so I boot to a blue screen. 
I tried printing the TSEC keys on hekate v4.2 but I'm getting the error: 

Unknown package1 version for reading
TSEC firmware (=  '20181107105733')

What should I do from here?

fennectech, 2018-11-24:
Without boot1 boot0 your screwed.  Go to a previous backup.  I beleve boot0
and boot1 are device specific.  If not please let me know @ctcaer and ill
arrange to get them a copy of 6.1.0 boot.bin

pjh4, 2018-11-24:
I unfortunately do not have a boot0 or boot1 file since I did not bother saving them. If it is possible to use another device's boot0/1 file, I'll gladly try it. 

Thanks for your help.

fennectech, 2018-11-24:
> Thanks again for your help and have a happy thanksgiving/birthday.

Thanks. Ill be heading home tonight.  Ill see if i can get someone to
powercycle my server and if i can ill toss the file up onto my mega.

CTCaer, 2018-11-24:
They are not, as a whole.

Choi desktop can be used to only flash the part that is not console unique and restore them to a previous version.

Also here is not a support forum. The issues sections on repositories, is not for that job.

closed 2018-11-24 by CTCaer

fennectech, 2018-11-26:
my apologies.   @pjh4  Email me at fennectech@gmail.com.  I cant share that file here.