CTCaer/hekate #153: 6.2 support

Nekxes, 2018-11-30:
Hey !
Since Atmosphère is now in version 0.8.1 wich support 6.2 and since Kosmos is only waiting for you to be compatible with 6.2 as well (https://github.com/AtlasNX/Kosmos/issues/158) do you plan to update Hetake soon ?

W34P0n-X, 2018-12-01:
@Nekxes [HEY!](https://www.youtube.com/watch?v=2kXMcjKhBgU)llo!

balika011, 2018-12-01:
We have no plans to doing so.

GamingKingKKu, 2018-12-03:
are you sure you are related to Hekate?

fennectech, 2018-12-03:
He seems to contribute a lot to hekate.

fennectech, 2018-12-03:
From the look of that PR.  6.2 work is already in progress

balika011, 2018-12-03:
What kind of PR?

fennectech, 2018-12-03:
Ack.  Nvm

balika011, 2018-12-03:
Tell me, fox. I'm curious

fennectech, 2018-12-03:
Was looking at the kosmos issue posted here.

balika011, 2018-12-06:
> 
> 
> I have no plans to doing so.

Because I didn't plan it at the time. It was almost done.

fennectech, 2018-12-11:
@GamingKingKKu you tease.

fennectech, 2018-12-11:
@balika011 You tease xD

m4xw, 2018-12-11:
Please @fennectech this is not a Chat

Rickardo987, 2018-12-11:
@fennectech oof