CTCaer/hekate #156: Update.bin not loading properly with Hekate 4.2 [Black Screen]

16BitWonder, 2018-12-07:
I was using a dongle with Hekate 4.2 flashed and extracted the latest release to my SD and placed the new Hekate 4.5 in bootloader/update.bin.

Every time the 4.2 payload was pushed it would result in a black screen when update.bin was present, removing it allowed 4.2 to boot normally.
I then flashed 4.5 to my dongle to avoid using update.bin and it's working fine that way.

I'm not sure how relevant of an issue this is considering it can be fixed by just flashing the newer payload, but figured to put this up in case others were having the same problem.

CTCaer, 2018-12-07:
I managed to check it again (on exfat and fat32 cards) and it works as intended.
Can you overwrite the update.bin with v4.5 again?

16BitWonder, 2018-12-07:
Put 4.5 back in bootloader/update.bin and reflashed my dongle with 4.2 and it's working now.
I'm not sure what, but I must have done something wrong last night. Sorry!

CTCaer, 2018-12-07:
You use exFAT?

Maybe a homebrew is corrupting your sd card and takes hekate with it.
Even if one bit changed and it's placed where it shouldn't the boot will hang.

I'll investigate more.

16BitWonder, 2018-12-07:
My SD is formatted as fat32

Pushing 4.5 and then shutting down from RCM let 4.2->4.5 work again.
Trying to push 4.2->4.5 after rebooting from HOS (6.2.0 Atmo-v0.8.1) 4.2->4.5 results in a black screen. (Including after shut down)
Pushing 4.5 and shutting down from RCM this time did not let 4.2->4.5 work again.

I'm still trying to figure out a good way of getting 4.2->4.5 working again after the black-screens start up.

16BitWonder, 2018-12-07:
I'm not sure if this will work 100% of the time, but I got 4.2->4.5 booting again by:
Moving update.bin to root
Booting Hekate 4.2
Tried to boot stock (failed, since I'm on 6.2.0)
Shutdown from Hekate 4.2
Moved update.bin back to bootloader\
Booted Hekate 4.2->4.5

After I got it working I tried launching Atmosphere with Hekate rather than Fusee.
-I booted into and rebooted from Hekate-Atmosphere three times with no issues so far
-I then tried booting into and rebooting from Fusee-Atmosphere (this is what I had been doing prior to this) and it's now black-screening again.

CTCaer, 2018-12-07:
Do you see hekate's logo before the black screen?
If yes, that's Auto HOS power off.
You can disable it in Options.

CTCaer, 2018-12-08:
Try what fails for you by injecting this special 4.2 
[hekate_ctcaer_4.23.zip](https://github.com/CTCaer/hekate/files/2659253/hekate_ctcaer_4.23.zip)

16BitWonder, 2018-12-08:
Pushed it a few times, still displayed a black-screen and had to be powered off manually.
I additionally tested a different exFat recovery SD I had to make sure it's not a faulty card and it's also black-screening with 4.2 and 4.23.

CTCaer, 2018-12-08:
I don't think it's the sd card or the copy of update.bin.
I had it my self after letting the device sit for some minutes in rcm (which is not consistent, most of the times it loads).
Otherwise it never happens. And it happens consistently for you, so I suspect diffs in hw.

Today I'll try to debug it. Can you go to RetroNX discord and hit me with a msg?

CTCaer, 2018-12-08:
Try with this update bin:
[update.zip](https://github.com/CTCaer/hekate/files/2659835/update.zip)

Based on tests I did, it seems that the Display controller fails to init.

If you remove autoboot, you'll see that the menu works. With autoboot on, it halts at last stage because it tries to deinit display again even though it is in a deinitialized state.

Our different experiences may also come because different DSI logic revisions.

16BitWonder, 2018-12-08:
After rebooting with the new update.bin from Hekate->Fusee->Atmospherev0.8.2 it continues to black-screen, sorry I missed your earlier message. I'll get on Discord.

CTCaer, 2018-12-08:
Based on testing with 16BitWonder, it seems that this happens only when fusee is used.
Full atmo with exosphere does not produce that.

It will have to wait for now, because no time to investigate more.

CTCaer, 2018-12-16:
https://github.com/CTCaer/hekate/commit/98d715243a3d806a25ca06c1260d9ea1f7c62d83

Issue https://github.com/CTCaer/hekate/issues/160 is also the same.

Thank you all for testing.

CTCaer, 2018-12-16:
**This is a breaking change.**
That means that your dongle/modchip must be updated with **the release that will include that fix**.

closed 2018-12-16 by CTCaer