CTCaer/hekate #157: Fatsfs error 13

Klauserus, 2018-12-07:
Hi. At first. Thanks for your work on hekate. My issue: i change my sd card to a samsung evo 256Gb U3. I format the sd card with different formatter into fat32. Copy my old files into it. When i start hekate 4.2 or 4.5 it shows an error message "...fatfs error 13...cant init sd card". I habe read about 1bit mode, but i think the card sticks correctly into slot. Do you have an idea?

Switch is in 5.1 and atmosphere 7.5. in card.

CTCaer, 2018-12-07:
If the problem is not the connector then it's the MBR.

I saw 2 persons having this problem.
And it was fixed by the following steps:
step 1: init the sd card (delete partition and wipe MBR)
step 2: create a new MBR (**not** GPT)
step 3: create a new primary partition.

The problem was that the MBR was missing some vital info that it was described as a MUST in the spec.
And FatFS requires that info to work.

Klauserus, 2018-12-07:
Hi
Great! It Works. Thanks man!
i had to do:  (windows) cmd -> diskpart -> select disk -> clean -> with easeUS Partition (free) FAT32 new primary partition (MBR)

have a nice evening
close

closed 2018-12-07 by Klauserus

lgmerinos, 2019-04-12:
Adding additional information for indexing purposes. Finding this answer took me almost a day. When upgrading from a 128gb sandisk sd to a 256gb sandisk sd xc after attempting to boot ReiNX i got: "Failed to init sd card"

I can confirm that @Klauserus solution works.

CMD -> diskpart -> cleandisk

And then on EaseUS Partition Free -> Format FAT32 as Primary Partition. 

Thanks a lot for the solution once again.

RogerCarvalhoBR, 2022-07-31:
Using Guiformat I found that for my 1TB card to be recognized correctly in Hekate and not have this error, the allocation unit size has to be **65536**.
[image]

impeeza, 2022-07-31:
yeap, big cards need 64K clusters.

CTCaer, 2022-07-31:
@RogerCarvalhoBR 

You should never use guiformat. It's fully bugged and does wrong formatting also.
It's a tool more useful for floppy disks and 2 decades old media.

Always leave whatever it came with or use sd association formatter if it has no filesystem.
And then use hekate to format it properly.
There's 0 need to do it before hand/2 times.