CTCaer/hekate #168: Commit 949a0345 does not actually work.

SciresM, 2018-12-17:
The Switch's bootrom checks an RSA signature over warmboot.bin. Simply patching it will not cause sleep to work -- this was either not tested, or tested incorrectly (not allowing enough time for the console to actually enter LP0 before pressing the power button again).

Editing warmboot.bin requires using a warmboot bootrom exploit, and I'd certainly hope people know better than to release one of those before the Mariko SoC is out. I don't see a warmboot bootrom exploit in hekate, so...

CTCaer, 2018-12-17:
Did you try it? ^^

closed 2018-12-17 by CTCaer

DavidBuchanan314, 2018-12-17:
Anyone want to explain how/why this works?

SciresM, 2018-12-17:
Yeah, https://github.com/CTCaer/hekate/commit/8b8f3c564c686db6e4ed7210114547c70d8a2fde exploits the bootrom to disable warmboot signature checks.

ZachyCatGames, 2018-12-17:
rip deja vu