CTCaer/hekate #169: PKG2 decryption failed

chardido, 2018-12-17:
What's about this problem? There is something I can do?

CTCaer, 2018-12-17:
Because this happens all the time, it seems like pkg1 and pkg2 do not match on version.
Did you restore anything that created that mismatch?

CTCaer, 2018-12-18:
The pkg2 resides on rawnand.bin.

When you did that you probably missmatched the packages.
So now pkg1 and pkg2 are from different firmware versions.

OR pkg2 is corrupt

chardido, 2018-12-18:
Thank you @CTCaer for reply.

I've already tried to reinstall 6.2 firmware with choidujour, but I have problem retrieving keys with HACGUI, so I'm stuck at this point (without keys I can't run choidujour).

lugede, 2019-12-26:
I met the same issue,  I update the atmosphere to 0.10.1, problem resolved.