CTCaer/hekate #172: Black screen after Nintendo logo when selecting CFW

omegaIuI, 2018-12-28:
I am unable to boot into Atmosphere via Hekate. If I try to do so, there is text output that appears as if things are loading correctly (I see no errors and I saw something green that had the word boot at the end), and the Nintendo logo appears, but I get a black screen indefinitely (I've tried letting it sit for 20+m min, no change) afterwards, right before the Switch logo.

I'm using Kosmos, and I've tried just about everything I and the people who were helping me on the Kosmos Discord could think of including:

- Quick reformatting SD card (both FAT32 and exFAT)
- Full reformatting SD card (FAT32)
- Doing a surface test on SD card (0 read errors found)
- Trying multiple versions of both Hekate and Kosmos (v4.5, v4.6 // v11.5, v11.7)
- Trying multiple versions of Atmosphere
- Updating 6.2.0 (orig. installed with system update prompt) > 6.2.0 (installed with ChoiDujoirNX, exFAT support enabled)
- Trying with AutoRCM off/on
- Trying with auto boot off/on
- Unplugging USB from Switch as soon as Hekate is injected (I am using a computer to inject payloads)

I'm not sure why this is happening, and nobody I've asked has an explanation so I am posting this issue as I think it _might_ be related to Hekate. fusee_primary boots into Atmosphere normally, but I am unable to with Hekate.

CTCaer, 2018-12-28:
What's your config? (boot entry)

omegaIuI, 2018-12-28:
> What's your config? (boot entry)
Not exactly sure what you mean, do you mean hekate_ipl.ini? If so, it's:
`[config]
autoboot=0
autoboot_list=0
bootwait=5
customlogo=1
verification=2
backlight=100
autohosoff=0

{AtlasNX/Kosmos v11.7}
{ }
{Discord: https://discord.gg/qbRAuy7}
{ }

{-- Custom Firmwares --}
[CFW]
kip1=modules/required/loader.kip
kip1=modules/required/pm.kip
kip1=modules/required/sm.kip
kip1=modules/required/fs_mitm.kip
secmon=modules/required/exosphere.bin
kip1patch=nosigchk
atmosphere=1
{ }

{---- Miscellaneous ---}
[Stock]
{ }
`

CTCaer, 2018-12-29:
Your ini seems legit.
Can you take a photo of Console info -> Sd card?

omegaIuI, 2018-12-29:
> Your ini seems legit.
> Can you take a photo of Console info -> Sd card?

Sure, here yo go:
[image]

CTCaer, 2018-12-29:
Hmm your sd card seems like a legit sandisk A1. So it shouldn't be a fake card problem.
So it shows the nintendo logo and not the switch logo? Or it shows both?

omegaIuI, 2018-12-29:
> Hmm your sd card seems like a legit sandisk A1. So it shouldn't be a fake card problem.
> So it shows the nintendo logo and not the switch logo? Or it shows both?

Yeah I bought it at Walmart. And it only shows the Nintendo logo, it does not show the Switch logo.

CTCaer, 2018-12-29:
Can you add `debugmode=1` in your config and try?
```
[CFW]
kip1=modules/required/loader.kip
kip1=modules/required/pm.kip
kip1=modules/required/sm.kip
kip1=modules/required/fs_mitm.kip
secmon=modules/required/exosphere.bin
kip1patch=nosigchk
atmosphere=1
debugmode=1
```

omegaIuI, 2018-12-29:
Yeah, same thing, black screen after Nintendo logo, and I left it on until just now. No change.

CTCaer, 2018-12-29:
I'll have some time tomorrow to test some things.
I'll send you a msg when it's possible

omegaIuI, 2018-12-29:
Alright, I am probably going to be out after 12:00pm EST until probably 10-11pm, but I should still probably be able to reply to messages and possibly do things with the Switch. I appreciate the help a lot :]

knox, 2019-01-10:
I wonder whats the current status of this because i am having exactly the same issue

dogepotion, 2019-01-10:
> I wonder whats the current status of this because i am having exactly the same issue

wow, nice username

also, it hasn't been resolved, its a very odd issue that nobody else seems (or I guess seemed now) to have until I presented it. CTCaer has been working a lot on trying to figure out what this is and nothing's been determined as a cause yet

with that being said, while a bit redundant to ask, have you tried *everything* I originally listed that I tried in the OP? and it's a black screen after the Nintendo logo in Hekate? Fusee works fine?

knox, 2019-01-10:
Well, thanks for your fast reply.

To be honest, it's my very first time trying to boot a CFW at all. So i did not try out all the details you have mentioned before yet. Basically, i've read a ton of tutorials before i carefully backed up my device and then used sdsetup to create a zip which i'm currently trying to boot. After selecting "Custom FW" in hekate 4.6 i get a nintendo logo and then a black screen. Same happens when i manually choose the payload.

Fusee itself (that refers to the base exploit itself, doesn't it?) works fine on my device. At least i have successfully bootet several times into a bunch of different payloads.

Good hint regarding the usage of macOS. Indeed thats my current OS and obviously that's not the best choice for creating a sd for the switch. I'll give Linux a try tomorrow and see how it goes.

Probably a github issue is not a good place to discuss this kind of newbie topics so i'm sorry for disturbing here. On the other hand, if there is anything i can do to help soring out this problem i'd be glad to help.

dogepotion, 2019-01-10:
There are a lot of good Discords and websites (gbatemp.net is a pretty good place to start) that can help with setting up and booting CFW.

My mistake, I should've been more descriptive- by fusee I meant the fusee-primary payload that comes with Atmosphere. It boots Atmosphere slightly differently than Hekate. Not sure if it will make a difference if it's OSX messing with the files on the SD card, but if you try it and it doesn't work it's probably a good indication that OSX is doing something. There are a bunch of people who can't use OSX to create SDs for the Switch at all, while some can. Not sure what's the difference between those who can't and can.

knox, 2019-01-10:
I'm well aware of gbatemp and those other resources, thank you ;-) I came across your issue because the symptomes are matching mine exactly. 

Manually selecting fusee_primary payload leads to the same results for me.

I'm a bit astounded what possibly could have gone wrong when i extraced the files using homebrewed 7z directly onto the sdcard but surely you're right with your hints macOS beeing a rather well known source of problems here.

CTCaer, 2019-01-11:
@knox Your sd card is probably exFAT and you are missing the micro update. Reformat it to FAT32 with 32KB cluster and try again.

knox, 2019-01-11:
@CTCaer Thank you very much! This solved my issue.

Bruce055, 2019-01-31:
Hello, I'm having the exact same problem. 
Tried everything in the OP and nothing seems to solve it.

The switch shows the nintendo logo, but not the switch logo.

Any idea on what may be causing it, or how can I fix it? downloaded everything (the SD content, the payload, Tegra, etc, from the SDsetup, so I assume everything is from the latest version).

One thing that I'm noticing is that, when i boot the OFW with the SD card on, a message is shown that says I should make an update to use a microSD. But since I'm already on version 6.2.0 I don't want to do it as the new version is not supported yet.

Any help would be greatly appreciated.

knox, 2019-01-31:
@bruce055 see this answer https://github.com/CTCaer/hekate/issues/172#issuecomment-453312793

Your SD card is in exFAT format. That is why OFW asks you to install the update. There are two solutions: either reformat the SD card with FAT32 or install the update.

Bruce055, 2019-01-31:
Damn, It was that... I misread when CTCaer answered you and though I should format for _exFAT_ with 32Gb Cluster ¬¬ 

Sorry for the trouble and thank you very much for the help!!

CalBear91, 2019-02-15:
I am having the same issue with the black screen after nintendo logo when booting CFW or OFW.. My sd card is formatted to FAT32. So Im not quite sure what the issue is? Any help much appreciated!

urherenow, 2019-02-16:
First, google “switch sdfiles”, get the latest package, and overwrite everything. If it still doesn’t work, ask on Discord or gbatemp

MrAttraction, 2019-02-17:
I'm having the same issue :/ everything was working normally, now when I inject fusee primary I get stuck at a black screen after Nintendo logo (when it shows up.. sometimes I get stuck after KIP patching screen goes by)

I'm on 6.2 and I have the exfat update, I was playing fine till now

akwiz, 2019-03-09:
Hey guys I know this is an old thread but I was having similar issues for a while. Do you have autoRCM enabled and do you have a jig in your switch? I was having the same problems but my switch would not boot with the jig inserted.

CTCaer, 2019-03-10:
OP had a very different problem. One that doesn't make sense.
Even the BIS keys didn't match. I assume that like how the keygen is now, it may work now.

Feel free dogepotion to reopen if hekate 4.8 and up didn't fix that.

closed 2019-03-10 by CTCaer

justusiv, 2019-05-28:
@akwiz did you ever see logos or after hekate menue it would just go black? Mine just goes black currently and will try your comment when i get home.

justusiv, 2019-05-29:
For anyone reading this later.... it appeared to be a corrupt internal nand. After a restore of a previous backup it is boot just fine.

yratof, 2019-09-24:
How did you restore the internalnand @justusiv ? A link to follow for other?

justusiv, 2019-09-24:
@yratof 
I had a backup of my nand. However any time i upgraded firmware i had this problem using any bootloader except the paid one :( . I tested the paid bootloader which i wont name, but it just worked so i just started using it. I have not gone back and tested with any newer versions since i posted about my issue.