CTCaer/hekate #174: kip conflicts ?

newskywalker7, 2019-01-07:
Hello,

I try to load sys-ftpd.kip + ldn_mitm.kip, but the console freeze at Switch logo.
Here are my custom.ini file

"
{----- ALL -----}
{Dont run SDFilesUpdater with this!}
kip1=modules/required/loader.kip
kip1=modules/required/pm.kip
kip1=modules/required/sm.kip
kip1=modules/required/fs_mitm.kip
kip1=modules/sysftpd/sys-ftpd.kip
kip1=modules/ldn_mitm/ldn_mitm.kip
secmon=modules/required/exosphere.bin
kip1patch=nosigchk
atmosphere=1
{ }
"

I don't change anything in hekate_ipl.ini, maybe I'm wrong ?

Thanks for your help !

CTCaer, 2019-01-07:
Probably an applet mem pool abuse.
This means that there's no applet memory left and hangs.

newskywalker7, 2019-01-07:
Thank you for your answer. When I remove one of the kips it works well.
On the other hand, I don't know at all what I have to do to lower the mem needed, but its'OK (I'm not a progammer).

Do yo know maybe where I can find some information / explanation about kips loadings restrictions ?

Thank for your great job !

CTCaer, 2019-01-07:
You need to ask the creators of these kips.
Some info on these memory pools, you can find in [switchbrew.org](https://switchbrew.org)

Also, have in mind that this falls out of the scope of hekate. Hekate only packs them for HOS to load them.

closed 2019-01-07 by CTCaer