CTCaer/hekate #175: Hekate only tries to load sleep driver on initial load

shchmue, 2019-01-13:
Is it possible to also try to load it again before calling (or during) `hos_launch` if `ERR_LIBSYS_LP0` is set or is there some reason it must be run first? Otherwise, if SD is inserted after payload is sent, everything will work normally until sleep forces reboot.

CTCaer, 2019-01-13:
That config is universal for all OSes, that's why it runs at main.
The sd card is supposed to be inserted before injection. I don't find a reason not to.
I mean, you can always do a reboot to rcm, if you forgot to plug your hb sd card.

(Sleep does not forces a reboot when lp0 is not configured. It simply hangs.)

shchmue, 2019-01-13:
Yeah, someone captured this edge case when they noticed sleep worked when chainloading fusee-primary but not hekate hos launch. Ultimately it's up to you whether you want to failproof it of course, but I'm not sure what that entails for other OSes.

CTCaer, 2019-01-13:
It's not edge case. He simply didn't copied the library. I can't fool proof that.
You are supposed to copy the bootloader folder.

It's like saying, for fusee primary to not hang because there's no sd card or secondary is missing.

shchmue, 2019-01-13:
The edge case being forgetting to insert SD until after pushing payload, not having the sleep library missing altogether.

smashbro73, 2019-01-14:
I captured that sleep driver issue when I accidentally forgot to insert SDcard before injection. I  have everything required files in bootloader folder.

CTCaer, 2019-01-14:
Again. This is layer8.
>But launching fusee-primary with hekate is fine on that case.

This is irrelevant and wrong testing. Try running fusee without sd card and tell me what happens.
You can't test one payload without sd card and the other with sd card and say that the first has an issue.

Anyway, this is intended behavior and that's why the error is there.
The reboot to rcm is there for that reason also.
I don't want to fill the code with ifs and elses. Additionally, that lib may become irrelevant in the feature.

closed 2019-01-14 by CTCaer