CTCaer/hekate #176: Get stuck at "Cannot Identify Package One" when trying to boot cfw on 6.2.0

therf311, 2019-01-13:
If you need the numbers after "cannot identify package 1" i can post them.

CTCaer, 2019-01-14:
Please use the latest version: v4.6

therf311, 2019-01-14:
I am.

CTCaer, 2019-01-14:
what's the pkg1 version?

therf311, 2019-01-14:
Sorry, what is that? Kind of a newbie here.

CTCaer, 2019-01-14:
The number after that message.

Btw the message "cannot identify package 1" does not exist in hekate. So what are you using?

CTCaer, 2019-01-14:
You use v3.2 and not the latest.
That message was change in v4.0 and it's now 
```
Unknown pkg1,
Version (= '20181107105733').
```

please inject the newest. 
https://github.com/CTCaer/hekate/releases/download/v4.6/hekate_ctcaer_4.6.zip

therf311, 2019-01-14:
So I injected the 4.6 version, and when I select the "Load Firmware" option, it displays "Failed to load /hekate_ipl.ini/. "Using default launch configuration." "Could not identify package 1 version <= /20181107105733/>."

CTCaer, 2019-01-15:
Hmm, load firmware? v3.2
What does it say above the menu?

Also how you inject the payload?

therf311, 2019-01-15:
Above the menu it say hekate - ipi.
I use Tegra Rcm GUI to inject it.
Am I using the wrong payload? Is that it?

CTCaer, 2019-01-15:
That payload is even older than v3.2.
You need to add it in the list of that GUI and select it to inject it. Currently you launch an older version

therf311, 2019-01-15:
Where can I get the newer version?

CTCaer, 2019-01-15:
https://github.com/CTCaer/hekate/releases/download/v4.6/hekate_ctcaer_4.6.zip

therf311, 2019-01-15:
Which file do I inject?

CTCaer, 2019-01-15:
hekate_ctcaer_4.6.bin

CTCaer, 2019-01-15:
and the bootloader folder you copy it inside your sd card

therf311, 2019-01-15:
Thanks a lot. Worked like a charm.

closed 2019-01-15 by therf311