CTCaer/hekate #179: Linux issues, using warmboot in atmosphere

GavinDarkglider, 2019-01-25:
If you warmboot from atmosphere into hekate and try to run lakka, you loose all video card acceleration. not sure where the problem is exactly, but it works fine if you launch hekate as a payload from RCM and not atmosphere.

natinusala, 2019-01-25:
Can you define "loose all video card acceleration" ?

GavinDarkglider, 2019-01-25:
OpenGL stops working. X11 still runs, GPU profiles cant be changed, Retroarch background moves like a snale, N64 is unplayable, and even NES games run slowish. CPU Profile still works though. It seems like coreboot isnt loading arm trusted firmware to enable the GPU capabilities, and possibly since boot time was much longer, is having issues with tegra_mtc blob as well, since atmosphere loads in minerva. I think the issue is that exophere is being used as the trusted firmware, which linux doesnt support. That being said, I dont have a way to dump the coreboot logs to see what is actually going on there(No UART adapter).

natinusala, 2019-01-25:
If OpenGL stopped working, you wouldn't see any background wave at all ;)

I will use my UART adapter to see what's going on, I'll come back to you once I have answers.

GavinDarkglider, 2019-01-25:
Mesa has a built in software openGL render. I am getting 5.6 - 6.3 FPS playing Zelda OOC in mupen64, and 5.5 in main retroarch menu. That being said, if it is keeping exophere in memory, and we can fix this issue, then it should also make sleep, shutdown, and reboot to atmosphere possible. :)

natinusala, 2019-01-25:
You can look at the RetroArch logs to see the GL renderer being used.

Exosphere isn't kept in memory, coreboot spins BL31 up when booting anyway

GavinDarkglider, 2019-01-25:
I am not sure what the problem is then, if exophere isnt left in the memory. From what I can tell, mesa is being used for an opengl render, but there there is only /sys/kernel/debug/dri/0/ and /sys/kernel/debug/dri/128/ when using atmosphere's warmboot feature, where when booting from RCM there is /sys/kernel/debug/dri/0/ and /sys/kernel/debug/dri/1/ and /sys/kernel/debug/dri/128/ and /sys/kernel/debug/dri/129/ and my framerate on Zelda is goes between 27 and 60 fps.

natinusala, 2019-01-26:
Okay, so UART output is also killed by rebooting from Atmosphère. I couldn't grab any log.

I confirm that all GPU features are broken when rebooting from Atmosphère.

For the record:

RetroArch logs:

```
[INFO] [recording] twitch streaming key empty
[INFO] RetroArch 1.7.5 (Git fcf75fda17)
[INFO] === Build =======================================
Capabilities: NEON ASIMD
Built: Jan 22 2019
[INFO] Version: 1.7.5
[INFO] Git: fcf75fda17
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/dri)
libEGL warning: DRI2: failed to open swrast (search paths /usr/lib/dri)
[ERROR] [EGL]: #0x3001, Unknown
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
Segmentation fault
```
The segmentation fault is due to the GL context being null.

Xorg logs: https://pastebin.com/g1tCMK66
dmesg: https://pastebin.com/HGRJWRKD

CTCaer, 2019-01-26:
Fixed internally.

This will either follow a commit to hekate (it may help with any possible HOS boots, if there are)
or coreboot.

We'll see.

CTCaer, 2019-02-11:
@GavinDarkglider
https://github.com/CTCaer/hekate/commit/4f2bbbf101f3fda0771685e0a40d6876005c2c36

closed 2019-02-11 by CTCaer