CTCaer/hekate #184: Hekate reboots to RCM right as I press on my CFW boot entry

ThatNerdyPikachu, 2019-02-13:
While using the latest master (v4.6 tag by this point) of hekate, I've experienced an odd issue in which it bootloops into RCM?

```
[config]
autoboot=1
bootwait=2
customlogo=1
verification=2
autohosoff=1
backlight=100

[Atmosphere]
atmosphere=1
kip1=cfw/*
kip1patch=nosigchk
secmon=cfw/exosphere.bin

[Stock]
```

Right after i select the ``Atmosphere`` option (or let the autoboot happen), Hekate logs like it normally does (all my patches were applied successfully, etc), and then reboots back into RCM.

ZachyCatGames, 2019-02-13:
Hekate doesn't work with recent commits of atmosphere as of right now

ThatNerdyPikachu, 2019-02-13:
thanks scires 🥖 

for reference, what commit in AMS broke it so I can roll back?

ZachyCatGames, 2019-02-13:
idk. I know that 0.8.3 stable works :/. It has something to do with exosphere though iirc

ThatNerdyPikachu, 2019-02-13:
has to be within this commit range then, if what you're saying is true.
https://i.imgur.com/doUHQgM.png

ZachyCatGames, 2019-02-13:
Probably?  ¯\_(ツ)_/¯

ThatNerdyPikachu, 2019-02-13:
Reverting just Exosphere to commit ``0e4c3007453df7787c25c9bc4d63d0858c658cdb`` fixed this, for now at least.

CTCaer, 2019-02-14:
Ah this? assuming you are using the new exo cfg mailbox (new magic and new addr 0x8000F000)
it's an exo bug.

Sec..

CTCaer, 2019-02-14:
https://github.com/Atmosphere-NX/Atmosphere/commit/46f4896992f08828ca964f94985e8e1b623e2133#r32317021

closed 2019-02-14 by CTCaer