CTCaer/hekate #185: SD Card and eMMC data (@lba 034ED000), do not match

Real96, 2019-02-17:
I'm trying to make Backup eMMC RAW GPP but it gave me this error:

```
SD card and eMMC data (@LBA 034ED000), do not match!

Verification failed...

Press any key and try again...

Time taken: 233m 43s.

```
I've got a 64GB Kingston SD and it's new, It stopped at the 90%

balika011, 2019-02-17:
Kingston cards known to be a lower quality. You have to retry the process. I never faced the same issue using Samsung evo cards.

Real96, 2019-02-17:
Going to try with another card and let you know what happens

CTCaer, 2019-02-17:
Take a readable photo of `Console info -> Print Sd card info`
and up it here

Real96, 2019-02-17:
Unfortunately i've got only Kingston 16GB SD cards, so i would like first to know if the problem can be solved before i spend money on a new SD card....Here the screen you asked:
https://imgur.com/JthbY9s

Real96, 2019-02-17:
If needed i've also got the screen of the error, but i wrote everything in the issue

CTCaer, 2019-02-17:
Your card is fake :/

If you want to do the backup now, artificially limit it.
Using a partition manager, get it to 16GB.
(24 because it fails at 26GB, but it may have a good faking algorithm and just overwrites other stuff.)

The thing is that your kingston 16GB, if it's real, it will be better and faster than this fake one.

closed 2019-02-17 by CTCaer

Real96, 2019-02-17:
Is there a way to check if my cards are fake or not using this screen? How did you do?

CTCaer, 2019-02-17:
I check CID (like this screen)
Some have crap faking, others very good.

And then format it and test with hw2testw or another fake card tool tester.

They basically write data and then they read it back and compare it bit by bit.

Real96, 2019-02-17:
Dang i put the 64GB one near my 16GB card and it's so obvious that it is fake...you told me to limit to 16GB with a partition manager, do you mean the switch backup i made?

Real96, 2019-02-24:
Sorry for late replay, in the end using an original 64GB SD card worked perfectly. Thanks a lot for your reply! You made a great job!