CTCaer/hekate #187: Totally disable Hekate bootlogo

Diablos90, 2019-02-22:
Hi, I'm chainloading fusee-primary.bin with autoboot, and Atmosphere actually haves its own bootlogo, so I'm trying to disable the Hekate bootlogo (with no luck).

I read that you can disable the Hekate bootlogo by setting up "bootwait=0" on hekate_ipl.ini. 
I tried to do it, but the bootlogo appears anyway (for a fraction of a second).

Dont know if this is a issue or or I'm doing something wrong, I hope there's a way to totslly disable it.

Thanks in advance for all your work :)

CTCaer, 2019-02-24:
https://github.com/CTCaer/hekate/blob/master/bootloader/main.c#L1055

Now checks if bootwait is 0

Diablos90, 2019-02-24:
Yeah, now bootwait=0 disables the bootlogo, thank you :)

I also noticed that with this 4.8 the custom bootlogo wont load: I tried to set customlogo=1 and bootwait=2 on hekate_ipl.ini, and I placed bootlogo.bmp (one that works with 4.6svc) inside the bootloader folder, but it doesn't work.

CTCaer, 2019-02-24:
That's strange, it works for me.
https://github.com/CTCaer/hekate/blob/master/bootloader/main.c#L989

Also customlogo=# is obsolete now.

hekate will try to load your `logopath=` bmp, if it doesn't find it, it will load `bootlogo/bmp`, if not built in one.

Diablos90, 2019-02-24:
I tried 2 bootlogos that works with 4.6svc and removed customlogo=1, sadly with no luck.

These are the bootlogos I tried:
[bootlogo 1](https://gbatemp.net/attachments/bootlogo-bmp.137745/) and [bootlogo 2](https://gbatemp.net/attachments/bootlogo-bmp.138033/)
(tried one at a time and named bootlogo.bmp).

This is my hekate_ipl.ini:
[config]
autoboot=1
autoboot_list=0
bootwait=2
verification=2
backlight=100
autohosoff=0
autonogc=0

{-- Custom Firmwares --}
[Atmosphere]
payload=bootloader/payloads/fusee-primary.bin
{ }

[ReiNX]
payload=bootloader/payloads/ReiNX.bin
{ }

{---- Miscellaneous ---}
[Stock]
{ }

CTCaer, 2019-02-24:
It's important if you boot atmo via hekate or using a modchip.

I'll check tomorrow for that.
for now you can use `logopath=pathto.bmp`

Diablos90, 2019-02-24:
Thanks again :)

mattytrog, 2019-02-26:
All is working here.
Removed the dimensions (720, 1280) from the bmp sanity check. Left B, M & 32 in.

This happens when x & y axis are switched. For example, if you are using a logo meant for ArgonNX.

Removing this part of the sanity check will make a switched axis bmp not display properly, but at least it won`t leave you scratching your head.

gfx_render_bmp_argb requires modification to automagically flip incorrect axis IMHO

CTCaer, 2019-02-26:
@mattytrog what you proposed is irrelevant and also prone to heap corruption.

This stopped working for another reason. Which from what I see it doesn't make sense.
The only case where the default logo is shown, is when there's no bootlogo.bmp or logpath, or bmp is not correct. (Bigger x or y size, not 32bit, etc)

I suppose that something is messing with https://github.com/CTCaer/hekate/blob/master/bootloader/main.c#L980

mattytrog, 2019-02-26:
Fair enough! I`m very C rusty. As default, with menu_logos etc in. It compiles to 127000 ish. Bigger than 126296.

I`ve tried lots of bootlogos and as I wrote above, this "seems" to be the reason.

Or maybe there is a little mem corruption somewhere on account of the compiled size being "too big"?

In my Simple-UF2 abomination, menu_logo blz is removed and the bootlogo blz is smaller I think (IIRC).

With the extra bits I`ve butchered and bolted on, compiled size still only comes out to 116000 ish.

Thanks for the great software! And BTW. Yep. From temp ;)

CTCaer, 2019-02-26:
The size is because you have older GCC. New version fixed a bug it introduced, that was saving debug info.
What you see in releases, it's from that compiler.

have in mind that if the payload size is more than 126296, it fails to relocate, and you stay with a black screen.

mattytrog, 2019-02-27:
That makes sense. I`ll update GCC. Your default Hekate should compile under 126296 once I have done that.

My modified version should be even smaller than 116000.

I also have "native" USB working in a test build I made. However I had to chop a load of stuff out to get the code to fit.

On with the project. I hope you don`t mind me making my own mods for it!

Thank-you once again for your work!

m4xw, 2019-02-27:
@mattytrog You should check out our module system for USB, I assume with "native" u mean using bootrom?
We don't mind forks, but we encourage ppl to contribute to upstream (aka this repo)

mattytrog, 2019-02-27:
Certainly I will.

I feel quite shy about my code. Always have done! But of course I`ll push back! I don`t want to be considered "rude" or anything!

Thank-you.

CTCaer, 2019-03-04:
@Diablos90 
So, it's a stack corruption, introduced in 4.8.
Hopefully it only affects /bootloader/bootlogo.bmp.
Next version will have the fix.

Diablos90, 2019-03-04:
Thank you :)
I found another anomaly, it's better to open a new issue?

CTCaer, 2019-03-04:
@Diablos90 what is ti about?

Diablos90, 2019-03-04:
You've probably already noticed it: navigating through the Hekate menu (for example going on Launch and coming back) cause a distortion to the red/white CTCaer logo on the bottor right (if you repeat Launch -> Back many times the distortion worsens).

CTCaer, 2019-03-04:
With or without update.bin?
sept run?
auto boot and VOL-?

Diablos90, 2019-03-04:
With update.bin inside bootloader folder.
Sept inside the sept folder is the Atmosphere standard one.
Yes, autoboot enabled and VOL- to enter Hekate menu.

CTCaer, 2019-03-04:
Can't replicate it at all locally. It seems it was fixed as a side effect of fixing the stack corruption.

Diablos90, 2019-03-04:
Perfect, I'll check again when the new version will be released, thanks again :)

CTCaer, 2019-03-08:
All these are fixed via https://github.com/CTCaer/hekate/commit/0ddc1c71a8689fa364f5838546b1b63a1d1be53e#diff-b6be68663e95ac400d2c531258deea2bR127

closed 2019-03-08 by CTCaer