CTCaer/hekate #189: Hekate still auto-reboots into RCM on boot entry selection

ThatNerdyPikachu, 2019-02-24:
```ini
[config]
autoboot=1
bootwait=2
customlogo=1
verification=2
autohosoff=1
backlight=100

[Atmosphere]
atmosphere=1
kip1=cfw/*
kip1patch=nosigchk
secmon=cfw/exosphere.bin
warmboot=cfw/lp0fw.bin

[Stock]
atmosphere=1
secmon=cfw/exosphere.bin
warmboot=cfw/lp0fw.bin
```

Fresh clones of both Hekate and Atmosphere, no clue what to say...

CTCaer, 2019-02-24:
update your update.bin and sept/payload with hekate4.8
And make sure lp0fw is correct

Also you are sure that is RCM?

ThatNerdyPikachu, 2019-02-24:
I'm booting on 6.2.0 for now, to update to 7.2.0
and yes, my payload injector reinjects after hekate displays green Booting... text

urherenow, 2019-02-24:
You’re using an auto-injector? Try turning that feature off.

ThatNerdyPikachu, 2019-02-24:
No, I left my injector plugged in.
I just tried unplugging it, screen went black after Booting...
after that, I plugged it into my PC and was able to use fusee launcher.

definitely RCM

ThatNerdyPikachu, 2019-02-24:
it works on AMS profile -- not on Stock tho
does Exo / lp0fw require AMS sysmodules?

CTCaer, 2019-02-24:
```ini
[Stock 7.x.x]
secmon=path/to/exosphere.bin
warmboot=path/to/lp0fw.bin
atmosphere=1
```

ThatNerdyPikachu, 2019-02-25:
i'm using a profile that is exactly the same -- see above ini

CTCaer, 2019-02-25:
It should work then. Try to add a newline after `warmboot=cfw/lp0fw.bin`

Rickardo987, 2019-02-25:
Is “semi-stock” safe enough to go online with? ~~BuT RiCkArD0!?, nOtHiNg Is 100% SaFe?!!?1~~ Not 100% safe but probably safe?

ZachyCatGames, 2019-02-25:
> Is “semi-stock” safe enough to go online with? BuT RiCkArD0!?, nOtHiNg Is 100% SaFe?!!?1 Not 100% safe but probably safe?

I’ve been using atmosphere online every day since hekate got sleep support :P. I’m not banned. So just exo and lp0fw is probably also pretty safe

CTCaer, 2019-02-25:
Nothing is 100% safe.
But `pretty safe`? Yeah.

CTCaer, 2019-02-26:
@ThatNerdyPikachu so you fixed it?

ThatNerdyPikachu, 2019-02-26:
@CTCaer haven't been able to test -- been sick for the past 3 days

will close

closed 2019-02-26 by ThatNerdyPikachu

CTCaer, 2019-02-26:
Then forget the testing, get well.