CTCaer/hekate #190: Hekate fails to boot after ejecting / inserting the sd card when in the main menu.

ITotalJustice, 2019-02-25:
Hekate 4.8
Cfw = kosmos
FW = 6.2

Whilst in the hekate main menu, removing the sd card and re inserting it will cause hekate to hang on the booting text when selecting cfw.

A workaround is after re inserting the sd card is to open launch, exit back to the main menu, then re enter launch which will allow it to boot cfw. In rare cases, doing this can result a error when booting, retrying after seems to work.

Haven't tested chainloading or booting stock.

[hekate_ipl.txt](https://github.com/CTCaer/hekate/files/2900498/hekate_ipl.txt)

urherenow, 2019-02-25:
When has arbitrarily removing an sd card with the power on been an acceptable practice on any OS, ever? 

You are essentially booting from it. Would you yank the cable from your root or C:\ drive and expect the system to function?

ITotalJustice, 2019-02-25:
Not sure what you're talking about. It's not the same thing at all.
This has been a feature of hekate since the first release, also when in the hekate menu the sd card is classed as unmounted and therefor casues no harm to the sd card. Opening launch attempts to mount the sd card.

urherenow, 2019-02-25:
I read “main menu” in the title and was thinking you were in the Switch’s main menu 🤦‍♂️. Don’t mind me... I’m half-asleep.

ITotalJustice, 2019-02-25:
> 
> 
> I read “main menu” in the title and was thinking you were in the Switch’s main menu 🤦‍♂️. Don’t mind me... I’m half-asleep.

Not a problem, I editing my comment as I thought that might've been the case ;)

CTCaer, 2019-02-25:
Ye. Hekate was designed on some commits I did back in june to work like that.
Basically allowing you to eject on every menu (except launch and when in change option.)

Krude, 2019-02-28:
Just came here to note this when i saw someone else already mentioned the issue.

I'm on AMS 0.8.4 7.0.1 and have the same problem. hekate freezes/stalls on the green "Booting..." line.

CTCaer, 2019-03-08:
This was rectified in latest commits

closed 2019-03-08 by CTCaer