CTCaer/hekate #191: dump package1/2 in 7.0.1 not works

Ninoh-FOX, 2019-02-26:
when I try dump de package1/2, I see the menu that say that the console go to reboot for can read the new keys, but always reboot in atmosphere 0.8.4. ¿¿??

balika011, 2019-02-26:
pk11 dumping is not supported and not planned. pk2 needs a fix.

CTCaer, 2019-02-26:
Well currently I designed it to not touch /sept/payload.bin if it's not hekate.
I may add a logc where it is checking it, rename it, copy hekate, and reload that.
And then on launch, it will check if the renamed file exists and rename it back..

Ninoh-FOX, 2019-02-26:
yeah, now it works, thanks

CTCaer, 2019-02-26:
Well I still want to mitigate that.

CTCaer, 2019-03-04:
So I made some changes internally.
Next version will replace the payload if it's not hekate and it will also keep a backup of the foreign one.
And then restore it automatically.

That way if you have another preferred bootloader it will not mess with dumping pkgs or booting another hekate boot entry - profile. And on the other hand, hekate will not mess with your choices.

CTCaer, 2019-03-08:
https://github.com/CTCaer/hekate/commit/2e9a89aa20ec18baa8a715a57cabe59aef8b4be4#diff-521b6d2bc026ec90788825021df95568R150

So this now will rename, the foreign payload and write hekate in its place.

The need for update.bin is gone now. Hekate copies itself from IRAM

closed 2019-03-08 by CTCaer

Ninoh-FOX, 2019-03-08:
Great!! After of dump the file, Remplace this again the original file in the procces?