CTCaer/hekate #193: How to turn off the information output of logo and initializing....

love768952693, 2019-02-27:
Hello, developer. I want to know how to turn off logo output and initializing... information output.

I want to configure whether logo will never be displayed, and the information shown by initializing....

After selecting the custom menu, Nintendo information will appear and then enter the system.

CTCaer, 2019-03-01:
I'm not sure I understand..
If you don't want to see logging, don't use Launch, and setup Auto Boot.
If you don't want to show auto boot logo, you can choose disable it from Options.

love768952693, 2019-03-01:
First of all, thank you developers, the tools provided.

I want to disable this
[image]
and this
[image]

My current startup is started from Launch, so I can choose the space. After selecting the machine, the above two images will appear. I don't know how to stop them. I hope I can output the first loaded information, and the boot image. Direct Nintendo logo, just enter the machine.

urherenow, 2019-03-01:
Github is not a support forum, and the sept splash screen has nothing to do with hekate. There will never be a way to disable it.

CTCaer, 2019-03-01:
Sept logo for 7.x.x and up is here to stay for a specific reason.

As for the "Initializing.." use
`Options` -> `Auto boot` -> Choose the CFW taht you launch from `Launch`.
Then go to
`Options` -> `Boot time delay` -> `0 seconds (Bootlogo disabled)` to disable hekate's logo.

That way it will only show Sept logo and then nintendo.

love768952693, 2019-03-01:
The number is 10,000.

Thank you

CTCaer, 2019-03-01:
what number?

love768952693, 2019-03-03:
The meaning of this number is that I said thank you N times.

CTCaer, 2019-03-03:
Ah, you are welcome!

closed 2019-03-03 by CTCaer