CTCaer/hekate #194: hekate hangs at initializingfound kg1 (numbers) identified kg1 and keyblob 6 loaded pkg1and keyblob

Zedris, 2019-03-01:
Hey I just updated to kosmos files with sigs and tried hekate on 6.2.0 FW and every time i try to boot atmo im getting
Initializing...
Found pkg1 ("2018...........)
Identified pkg1 and keyblob 6
Loaded pkg1 & keyblob.

I am able to load stock fw and SX OS i was running kosmos 11.9 with 4.6 svc hekate with no issues before this, I did not delete modules/atmo/ and bootloader at first when i tried to update to the latest kosmos but i did delete them after and got the same message. i used a clean sd with nothing on it and fresh installed kosmos and still same issue.

urherenow, 2019-03-01:
This isn’t the Kosmos page.

CTCaer, 2019-03-01:
That's a problem of your dongle initializing CCPLEX (normal CPU) without a need.
Ask them to fix it or replace their broken bootloader with hekate directly.

closed 2019-03-01 by CTCaer

Zedris, 2019-03-01:
ah okay. thank you ct!

PandaUnit-TWL, 2019-03-10:
Zedris how did this turn out? I've got the same issue and I don't think this dongles getting an update (or at least I don't know how).

Tegrasmash from pc with hekate or fusee-primary.bin with dongle load atmo just fine, but I prefer hekate through my dongle.

Zedris, 2019-03-10:
> Zedris how did this turn out? I've got the same issue and I don't think this dongles getting an update (for this specific problem [sx dongle. i like it, it's small]).
> 
> Tegrasmash from pc with hekate or fusee-primary.bin with dongle loads atmo just fine, but I prefer hekate through my dongle.
> 
> _**edit**_ Ok did some reading, so I really should just inject hekate directly, not chain-loading through the sx os boot.dat
> 
> Then why does fusee-primary.bin work through chain-loading?

i guess nothing came of it i tried 2 different sx os dongles since they shipped 2 to me originally instead of 1 and both are doing it even during the newer atmosphere/hekate versions. so i just went and bought a ns atmosphere and the other dongle on ali express. to be completely honest i feel like something with hekate is blocking it from loading or the dongle isnt capable of launching it anymore. idk

CTCaer, 2019-03-10:
https://gist.github.com/CTCaer/13c02c05daec9e674ba00ce5ac35f5be

[boot.zip](https://github.com/CTCaer/hekate/files/2949086/boot.zip)
^ v4.9.1

PandaUnit-TWL, 2019-03-10:
CTCaer, are you saying use this boot dat with your 4.9.1?

CTCaer, 2019-03-10:
that file is basically 4.9.1.

So for the future you either drop the new version as update.bin in /bootloader/
or just [rename it or change the stage2 name] and run the script.

Zedris, 2019-03-10:
ty CT