CTCaer/hekate #195: Put vol- autoboot override before sept

noahc3, 2019-03-07:
Using vol- to access the menu on 7.0+ is now fairly slow because sept always runs first. Having the vol- override available before sept, either always, as an option, or both available before and after sept, would make things a bit quicker.

CTCaer, 2019-03-07:
This was changed internally anyway.
The reason is not slowness of accessing menu, but a more important thing.
Warmboot breaks usb and I still haven't found what to comment or add on usbd init.
(forced usbd init currently hangs the console after a warmboot. I assume some PLL is down.)

CTCaer, 2019-03-07:
Have in mind that if I fix memloader, I'll retain the current order.
With the added change that VOL- cancels boot immediately and goes to menu.
It will still check for 0ms if you press on before injecting.

CTCaer, 2019-03-08:
https://github.com/CTCaer/hekate/commit/4f4df9ccd1e2a792105dadb9eabfa0b637663afc

closed 2019-03-08 by CTCaer