CTCaer/hekate #197: Store configuration in a seperate file so updating kosmos and such doesnt overwrite it.

fennectech, 2019-03-08:
Can we store configuration in a seperate file so that its not overwritten when updating kosmos ?  this would allow users to update without instantly loosing all their settings.  Howabout /bootloader/ini/settings.ini

CTCaer, 2019-03-08:
Copy hekate_ipl.ini already.
hekate will not become a mess because some are bored to do a simple thing.

If this bothers you so much, you can ask them to not touch that file, but only when it's needed.

closed 2019-03-08 by CTCaer

fennectech, 2019-03-08:
problem is this is distributed with every update even when they dont touch it.   is it possable to load settings like autoboot from another ini already?

CTCaer, 2019-03-08:
nope. and that's not an issue with hekate.

fennectech, 2019-03-08:
understandable   thank you.