CTCaer/hekate #198: [Feature Request] Ability to boot semi-stock via FSS0 on 7.x with an ini option

thedax, 2019-03-09:
The new FSS0 feature is really nice allowing for easier setups, but it would be excellent if it could also have an option to ignore loading the kips within fusee-secondary, for the purposes of booting semi-stock on 7.x.

I'd propose it to work like this:

```
[Stock 7.x]
fss0=/path/to/fusee-secondary.bin
fss0ignorekips=1
atmosphere=1
```

Where Hekate wouldn't process kips within fusee-secondary, and only grab the secmon & lp0fw from it.

Of course, the option name could be anything, it's just an example.

The main reason this would be useful is so that end-users who can't or won't compile Atmosphere don't have to go fetch loose files from Kosmos or buildbots that build Atmosphere. They would be able to just download the latest Atmosphere release, and then use the stock option as expected.

CTCaer, 2019-03-09:
I may rectify this in a hotfix version, where I'll also address another issue

CTCaer, 2019-03-09:
@thedax 
Please test this:
[iplctc.zip](https://github.com/CTCaer/hekate/files/2948547/iplctc.zip)

Make sure you deleted sept/payload.bin, because I didn't change the version yet.

What it does:
if `stock=1` is found, it will disable loading kips from fss0 and disable all patching.
If `fss0=` is not used, it will still disable kernel patching.

Additionally, if `fss0` is set, it will automatically set `atmosphere=1`.
```
[Stock 7.X.X (fss0)]
fss0={sd path}
stock=1

[Stock 7.X.X (non-fss0)]
secmon={sd path}
warmboot={sd path}
atmosphere=1
stock=1

[Stock (older FW)]
stock=1
```

CTCaer, 2019-03-09:
So I managed to merge `Stock 7.X.X (fss0)` and `Stock (older FW)`

On the **released** version the config
```
[Stock]
fss0={sd path}
stock=1
```
will check if >= 7.0.0 and if yes it will load exo+wb, if not, it will load nothing.
As long as the fss0 binary exists.

CTCaer, 2019-03-09:
https://github.com/CTCaer/hekate/commit/14c50ed7f8d8ee36a47d7268a653868d3109bd37

closed 2019-03-09 by CTCaer

thedax, 2019-03-10:
I'm sorry I couldn't test it in time for you, but thank you for adding this! It's very helpful.