CTCaer/hekate #200: sd and emmc data <@LBA 00800000>, do not match

farukotakeru, 2019-03-09:
When I try to perform a full eMMC NAND backup on my Switch it always gives me this error:

SD Card and eMMC data <@LBA 00800000>, do not match!
Verification failed..

Press any key and try again...

I am currently running on hekate 4.9 with my microSD formatted to exFAT. My Switch is currently on 7.0.1. My microSD is a 128GB SanDisk Ultra.

MichaelRebirth, 2019-03-09:
I got the same problem, tested with:
64gb sandisk extreme
256gb sandisk extreme
128gb sandisk ultra 
*all exFAT
2 switch unit, same error msg... (verify 1/2 all the same result)

farukotakeru, 2019-03-09:
> When I try to perform a full eMMC NAND backup on my Switch it always gives me this error:
>
> SD Card and eMMC data <@lba 00800000>, do not match!
> Verification failed..
>
> Press any key and try again... [...]

Below is the test results of H2testw
`Warning: Only 118911 of 118912 MByte tested.
Test finished without errors.
You can now delete the test files *.h2w or verify them again.
Writing speed: 13.4 MByte/s
Reading speed: 20.7 MByte/s
H2testw v1.4
`

hlc1209, 2019-03-09:
Same here.
Samsung 128Gb

bzennn, 2019-03-09:
Same problem. Tested with Samsung Evo 128 GB

farukotakeru, 2019-03-09:
> When I try to perform a full eMMC NAND backup on my Switch it always gives me this error:
> 
> SD Card and eMMC data <@lba 00800000>, do not match!
> Verification failed..
> 
> Press any key and try again...
> 
> I am currently running on hekate 4.9 with my microSD formatted to exFAT. My Switch is currently on 7.0.1. My microSD is a 128GB Kingston.

Tried with Hekate 4.8 and it worked fine. Downgrading to Hekate 4.8 till this issue is fixed

CTCaer, 2019-03-09:
Hmm at 4GB mark.
I'll check.

Btw you all stop at 0x**00800000**?

farukotakeru, 2019-03-09:
> Hmm at 4GB mark.
> I'll check.
> 
> Btw you all stop at 0x**00800000**?

I did but I'm not sure about the rest.

> I got the same problem, tested with:
> 64gb sandisk extreme
> 256gb sandisk extreme
> 128gb sandisk ultra
> *all exFAT (original card) ams 0.8.5 & 7.0.1
> 2 switch unit, same error msg... (verify 1/2 all the same result)

> Same here.
> Samsung 128Gb

> Same problem. Tested with Samsung Evo 128 GB

Guys?

CTCaer, 2019-03-09:
No problem I found the issue.
It's the classic 64bit problem.

Thank you all for reporting that.

CTCaer, 2019-03-09:
Please test this:
[iplctc.zip](https://github.com/CTCaer/hekate/files/2948529/iplctc.zip)

Report back if it reached more than ~9%.

farukotakeru, 2019-03-09:
> Please test this:
> [iplctc.zip](https://github.com/CTCaer/hekate/files/2948529/iplctc.zip)
> 
> Report back if it reached more than ~9% in verification (green bar).

This is the payload to be injected via Tegra?

CTCaer, 2019-03-09:
Yeah. Locally I still name them iplctc.bin to distinguish them from officially released binaries.
It's hekate though.

farukotakeru, 2019-03-09:
> Yeah. Locally I still name them iplctc.bin to distinguish them from officially released binaries.
> It's hekate though.

Ah okay, gonna run it now

farukotakeru, 2019-03-09:
[image]

Looks like it managed to get past 9%

CTCaer, 2019-03-09:
Ok, I'll cook up a hotfix version

CTCaer, 2019-03-09:
https://github.com/CTCaer/hekate/commit/61401d733e416c830b0ea9f23be87c1042afc758#diff-10557f14a06929c52a3feda4d5d68cecR90

leonardobsjr, 2019-03-09:
Good lord, i was freaking out because i've tried 3 nand backups before checking the issues. Having the exact same problem: same address, same everything. @CTCaer it is safe to use Hekate 4.8 for the backup procedure or it's better to wait for the fix to be deployed?

CTCaer, 2019-03-09:
ofc it's safe. don't wait for 4.9.1 if you need to do that now.
4.9.1 will come out after I made sure that the new stuff are good to go.
Don't wanna a 4.9.2

closed 2019-03-09 by CTCaer

jcklpe, 2020-11-29:
@CTCaer : I'm currently having this problem though the address for me is 0020000 instead of 0080000  
  
Any clues on what is the issue?

mattdehat, 2021-03-07:
i have this problem also 
[image]
[image]

impeeza, 2021-03-07:
Check your card is almost sure a fake or damaged one. Use h2test

mattdehat, 2021-03-08:
Many thanks for the quick reply. X2 corrupt cards. I really appreciate your time and knowledge. Many thanks again.