CTCaer/hekate #201: Progress bar misaligned - fix below

mattytrog, 2019-03-09:
Progress bar etc overwriting previous text making restore dialog look untidy.

in fe_emmc_tools.c...

Line 668 is 
gfx_con_setpos(&gfx_con, gfx_con.savedx, gfx_con.savedy);

Change to
gfx_con_getpos(&gfx_con, &gfx_con.savedx, &gfx_con.savedy);

to fix.
Cheers.

CTCaer, 2019-03-09:
The fix is a little different.
I meant to hide the check for split files info.
And only show it if total size does not match or if there's a missing file.

Thanks for telling me though. Fixed it internally

CTCaer, 2019-03-09:
https://github.com/CTCaer/hekate/commit/61401d733e416c830b0ea9f23be87c1042afc758#diff-10557f14a06929c52a3feda4d5d68cecR90
and the changes below

mattytrog, 2019-03-09:
Fix tested. All good! Thx.

Tried the multipart now. Very handy.

closed 2019-03-09 by mattytrog