CTCaer/hekate #203: Reboot to payload into chainloading memloader

ckurtz22, 2019-03-16:
When I try to use the newish reboot-to-payload functionality of Atmosphere to reboot into hekate, and then launch memloader via chainloading with Hekate, it will run fine, but my computer won't mount the SD card. It works fine chainloading if I launched hekate through rcm though.

fennectech, 2019-03-16:
This has something to do with sept i think. Ive seen people talking about sept breaking memloader.

CTCaer, 2019-03-16:
It has to do with warmboot. So both sept and reboot to payload are affected.
There's no fix for now

ckurtz22, 2019-03-16:
Yeah I'm not even using sept (haven't really touched my switch since before 7.0). 
Not that big a deal if it is not an easy fix, works well enough just unplugging my SD card in Hekate (when I have a card reader with me)

CTCaer, 2020-05-01:
new Nyx has UMS built in

closed 2020-05-01 by CTCaer