CTCaer/hekate #206: hekate doesn't load

genx31, 2019-04-09:
everytime i try to launch hekate in rcm tegra. It's succesful on my pc but on my switch i see the logo of kosmos like 10seconds. After that reboot to OFW nintendo. Getting desparete.
Plz some help

CTCaer, 2019-04-09:
What do you mean? Hekate doesn't randomly boot OFW, except if you tell it to.
Make sure that you are in OFW (like you can't access hbmenu. How to access it differs with the version of CFW you use).
Also that the profile you boot is not stock.

Also in a CFW you are not going to see a different home menu.

legoj15, 2019-04-09:
> Also in a CFW you are not going to see a different home menu.

After you see the kosmos screen, you will see the Nintendo logo. that is normal. Try opening the blue album applet, and see if you get to the homebrew menu since that is the default HB launcher app on atmosphere kosmos

spectralshadow514, 2019-04-09:
Don't forget that if you're using the atmosphere version of kosmos, that you now have to hold R on a title (game or otherwise) to open up the homebrew menu.

CTCaer, 2019-04-15:
From the luck of a reply from OP, I assume that this was the first cfw experience.

closed 2019-04-15 by CTCaer