CTCaer/hekate #207: Ability to chainload fusee-primary

ThatNerdyPikachu, 2019-04-09:
While using Hekate to chainload f-p, Hekate replaces sept/payload.bin with itself, and then f-p can’t be used since sept will chainload to Hekate.

CTCaer, 2019-04-09:
You sure that your sept payload was really correct?

ThatNerdyPikachu, 2019-04-09:
Hekate replaces it with itself, right?

(I haven't actually tested this :/, just got a few reports from people and decided to open an issue here.)

fennectech, 2019-04-09:
You can load fusee-secondary directly with Hekate now.

ThatNerdyPikachu, 2019-04-09:
I know that.

ThatNerdyPikachu, 2019-04-09:
Still, not being able to load f-p is a bug, considering how Hekate is a bootloader.

CTCaer, 2019-04-10:
- Hekate never runs sept code or touches /sept/payload.bin when a payload is chainloaded.
 https://github.com/CTCaer/hekate/blob/master/bootloader/main.c#L643
 https://github.com/CTCaer/hekate/blob/master/bootloader/main.c#L924

- When hekate is really booting HOS and another foreign payload is at /sept/payload.bin, it renames it, writes itself there and after sept was run, it deletes itself and renames the original payload back.
 https://github.com/CTCaer/hekate/blob/master/bootloader/hos/sept.c#L74

Hekate currently is the only bootloader that can always boot HOS itself, without forcing itself to the user.
And this is because of that micromanage.

So this is clearly a user error.

ThatNerdyPikachu, 2019-04-10:
Seems so, will check back with them.

closed 2019-04-10 by ThatNerdyPikachu