CTCaer/hekate #210: Hekate 4.10 SD card issues

crckmc, 2019-04-23:
Since updating my switch to 8.0.0 and using Hekate 4.10 I'm unable to boot from my transcend SD card (128gb A1). 

With fusee primary it boots from the problematic SD card.
Another SD card also works with hetake. 

I have formatted and checked the card for errors about ten times so I dont think this is a hardware issue with my card.

Any way for me to debug?

Related posts on reddit:

https://www.reddit.com/r/SwitchPirates/comments/bg2535/updated_to_kosmos_121_console_doesnt_boot_anymore/?utm_medium=android_app&utm_source=share

https://www.reddit.com/r/SwitchPirates/comments/bg8g6n/hekate_wont_read_sd/?utm_medium=android_app&utm_source=share

CTCaer, 2019-04-23:
https://github.com/CTCaer/hekate/commit/91606334c46a93f0d8cd8e94b01aab40412c73e6

> [sdmmc] Revert 204MHz sd device clock
> Again some Sandisk U1 cards do not behave at all at speeds like that (204MHz / 102MB/s).
> Revert back to 163.2MHz / 81.6MB/s.

This is a known issue and a hotfix release for these cards will be released today

crckmc, 2019-04-23:
Thank you, I didnt look at the latest commits...

closed 2019-04-23 by crckmc

CTCaer, 2019-04-23:
https://github.com/CTCaer/hekate/releases/tag/v4.10.1