CTCaer/hekate #211: SD Card Read errors

pcwizard7, 2019-04-23:
My switch gets sd read errors when 

1. boot hekate it would have black screen for while then kick into hekate
2. hekate a message "sleep module missing" and when i try to go into launch it has "sd card missing make sure it inserted and use default settings"
3. I can get the correct menu eventually 
4. I get the boot.dat for sxos until i i get it

After trying your new version with hotfix in it 4.10.1 it fix it for while but revert back to broken again

CTCaer, 2019-04-23:
You probably rebooted to 4.10.
Replace your reboot_payload also.

And format the sd card to FAT32 while you are at it.

pcwizard7, 2019-04-25:
it maybe a case of dirty sd card slot as i blew into the slot and it made it better but not perfect. I know this is off topic but is there anyway of clean slots without opening and take it out?

CTCaer, 2019-04-25:
Normally the problem stems from the connector and not the actual slot.

pcwizard7, 2019-04-25:
How would i go about cleaning the connector in the slot?

ErichMoraga, 2019-04-25:
If the connector and/or slot is dirty (rather than defective), you could try initially cleaning/blasting with can of compressed air (available at WalMart, and any computer store).

CTCaer, 2019-04-26:
Closing as this was not an issue with 4.10's normal 204MHz clock of UHS104.

closed 2019-04-26 by CTCaer