CTCaer/hekate #215: SD and eMMC data (@LBA 00120000), do not match (Hekate 4.10.1)

0Liam, 2019-04-25:
This keeps happening each time Hekate tries to verify that the NAND backup is correct. It stops at 1% and then throws this error.
SD card is a SanDisk Ultra U1 card formatted as FAT32. Verification mode is set to Sparse.

I've also had it say (@LBA 000F8000) as well, when having Hekate overwrite the previous NAND backup.

CTCaer, 2019-04-25:
Check your sd card with h2test tool.
Also before that, attach a photo here, of Console Info -> Print Sd card info

0Liam, 2019-04-26:
H2testw decided to error out at 17% so I decided not to do one of those. However, when I checked SD card info using hekate it listed the card as A0 instead of A1, and whatnot. I compared this to a real 64GB SD card that came from the same Amazon page and that said A1. The speeds were also very different, with the fake "200GB" card displaying speeds of 25mb/s, whereas the 64GB card displayed speeds of 104mb/s. This was very different from the advertised 100mb/s on the cards page.

Will be closing this now as I've been able to make a proper NAND backup using the 64GB card.

closed 2019-04-26 by 0Liam