CTCaer/hekate #217: "Battery Desync" Option Missing?

dothackjhe, 2019-04-25:
I seem to be having a very odd issue with my Switch involving its battery where it gets drastically drained after some time of being in sleep mode. I heard from GBATemp that it can be fixed by using Hekate to "de-sync" the battery via the "Tools" menu. Yet, using the latest version of Hekate, I cannot seem to find it as an option.

CTCaer, 2019-04-25:
What you need to do is to calibrate the percentage. so you need to do a 
100% -> 0% (after HOS powers off drain the remaining via hekate to 3.2V) -> 100%.

The battery "desync" fix is for another problem where switch shuts of at ~40%. And it's now applied on boot in order to save code space.

dothackjhe, 2019-04-26:
Got it, thanks.

closed 2019-04-26 by CTCaer

tambosi, 2020-01-01:
> What you need to do is to calibrate the percentage. so you need to do a
> 100% -> 0% (after HOS powers off drain the remaining via hekate to 3.2V) -> 100%.
> 
> The battery "desync" fix is for another problem where switch shuts of at ~40%. And it's now applied on boot in order to save code space.

I'm having the problem when my battery goes from 100% to 1% in 1-2 minutes. 
Can you explain in more details how to make: calibrate the percentage. so you need to do a
> 100% -> 0% (after HOS powers off drain the remaining via hekate to 3.2V) -> 100%?

dothackjhe, 2020-01-02:
@tambosi Are you actively using a custom theme? I think mine was partly caused by it.

tambosi, 2020-01-02:
@dothackjhe I'm not. I'm using Hekate 4.6. I'm updating to Hekate 5.1.1 and I will try again.

tambosi, 2020-01-02:
I update Hekate to 5.1.1 and the problem still happens.

tambosi, 2020-01-02:
@dothackjhe How do you solved the problem?

tambosi, 2020-01-06:
@dothackjhe, just to you knoew. I drained all the battery by let hekake open and after a full battery load, the problem was solved.
Thanks for the help.