CTCaer/hekate #218: Charging Switch in hekate over USB on Windows 10 causes "USB device malfunctioned" spam.

yuukieve, 2019-04-26:
Not a huge deal, but figured I should report this. Both the latest and last few versions of hekate that I've tried cause "The last USB device you connected to this computer malfunctioned" notifications endlessly every few minutes (can be more sporadic as well) when plugged into my PC. I'm running Windows 10 Enterprise x64, version 10.0.17763.404 (latest stable, not an Insider build). 

I am using the USB cable that came with the Switch Pro controller, in order to minimize issues with the fuel gauge IC and the improper implementation of USB-PD inherent to the Switch.

balika011, 2019-04-26:
The cable doesn't matter. The issue is in the broken protocol of the switch itself. The ic will burn even if you use the ninty cable.
This issue happens because Hekate doesn't handle any USB event.

CTCaer, 2019-04-26:
It's completely on your usb controller and drivers.
Mine do not exhibit this because they think the device is just not responding and they stop talking to it.
This can only happen if you connect it after hekate was run.

The usb charging misbehaves even in RCM, where I saw it drawing 750mA from a usb2.0 500mA limited port and 1200-1400mA from usb3.0. But this is not enough to damage the ic when connected to a PC.

As a sidenote, in 7.x and inside HOS and with latest asmedia usb3 drivers, switch gets into a constant charging/discharging cycle.

If you want to only charge it, just go into RCM and use its official charger. If you still want to use your PC, inject from PC and leave it there. (Except if your drivers continue to pester you with notifications.)

closed 2019-04-26 by CTCaer

yuukieve, 2019-04-29:
Thank you for the information. I had it plugged in at the time because I was doing a NAND backup.