CTCaer/hekate #221: Is this normal or my battery is desync

renatosk, 2019-05-08:
so my switch died for the same reason as this issue [https://github.com/CTCaer/hekate/issues/217](url) i let it charge again , booted horizon with 3% a let it do a full 100% , but after this i noticed something , theres a diference of 1.7% between hekate and horizon , for exemple on horizon is 93% while on hekate is showing 92.3% , i did what you recommended on the issue 217 , leting it die again and draining the remain on hekate , but again i booted and let it charge full 100% on horizon so maybe this is why it didnt sync ?

CTCaer, 2019-05-08:
Percentages in hekate are not 100% correct.

It doesn't restore your battery calibration and uses the POR configuration. So the percentage will always differ with HOS.

renatosk, 2019-05-08:
ok thanks

closed 2019-05-08 by renatosk