CTCaer/hekate #222: Umm I got a "Fatal Error (2)" while trying to restore my NAND

dorianpb, 2019-05-14:
It says:
[fatFS] Error: CCHK

CTCaer, 2019-05-14:
The file read from the f_read function, was pointing to sector 0 or 1 of sd card.
These sectors hold the Master Boot Record and ofc no files.

That means that either your card is corrupt or a fake one.

dorianpb, 2019-05-14:
Wow, you actually looked at this - I'm honored!
More importantly, I think my card is legit, so if I reformat and put the rawnand.bin on it again it should be good?

CTCaer, 2019-05-14:
It should.
Btw, exFAT is easy to corrupt when using homebrew in Switch.
So after restoring, it's worth to switch to FAT32.

For reference, the correct (most compatible) cluster sizes are:

- FAT32: 32KB
- exFAT: 128KB

No matter the size of the card/partition.
(except if partition is smaller than 2GB).

dorianpb, 2019-05-14:
Is it possible to restore a backup from FAT32?
The only reason I was on exFAT was because of the 4GB limit

CTCaer, 2019-05-14:
Yeah.
You can split it with a file splitter.
Split it to 1GB parts or 2GB or 3GB or whatever. Just don't go more than 99 parts.

Pattern is:
rawnand.bin.##
Where ## is: 00, 01, etc

dorianpb, 2019-05-15:
Alright I did what you said and my switch is now fine. Thanks for your help.

CTCaer, 2019-05-15:
you are welcome

closed 2019-05-15 by CTCaer