CTCaer/hekate #224: error cant open hekate_ipl.ini

majedcom, 2019-06-09:
i am running latest hekate .. when i backup nand etc everything is ok then i run it again to install cfw but i get the error message cant find or open hekate_ipl.ini eventhough it is inside the bootloader file .. what is wrong with it. am on ofw 7.0.1 and i want to update offline to 8.0.1

fennectech, 2019-06-09:
Filesystem corruption perhaps?  Does it open in notepad on your PC?

majedcom, 2019-06-09:
both are ok .. i even did recopy the file to the sd just incase .. same result

fennectech, 2019-06-09:
Thats just plain weird.

majedcom, 2019-06-09:
tbh i just booted sxos and did update from it

majedcom, 2019-06-09:
also one more thing that did solve the problem after i booted hekate_fspatches_v3 i did choose hekate_ctcaer_4.6.bin by accident and it booted the firmware also .. previously i never had to do that so maybe this is a new change i did not know about

CTCaer, 2019-06-18:
What version you used of hekate?

closed 2019-06-30 by CTCaer