CTCaer/hekate #225: Implement EmuMMC option for boot config

UnicornsOnLSD, 2019-06-16:
With EmuMMC out, being able to have a Hekate profile for EmuMMC and stock would be great. Also, an option to not mount the SD card in HOS would be nice for separating SysNAND from EmuMMC.

JulianHH, 2019-06-16:
I‘d first like to thank you for your dedication with this, CTCaer!

In addition to what OP posted, I‘d like to extend this feature request to be able to initialize EmuMMC with the current ofw NAND and also to be able to extend the SD to use the full capacity.

Thanks in advance!

Also, if I should make a complete new request, please forgive me, I‘ll then do it!

CTCaer, 2019-06-30:
https://github.com/CTCaer/hekate/commit/bd7f572989de35d5ae7188ba967da56ca76fe7de

closed 2019-06-30 by CTCaer