CTCaer/hekate #226: Update Hekate to support EmuMMC and Atomsphere 0.9.1

gr33k, 2019-06-20:
Update Hekate to support EmuMMC and Atomsphere 0.9.1

This will also allow Kosmos to update their build :)

Checked https://github.com/CTCaer/hekate/issues/225 but did not seems to be the same (trying to avoid a dupe!).

Thank you CTCaer for all your hard work and contributions to the open source community!

SwitchLover, 2019-06-26:
is this done yet or not

nicoboss, 2019-06-26:
> is this done yet or not

They already have it working privately and according to m4xw it will be released "soon(TM)".
NYX might end up being the successor of hekate but that's just speculation: https://twitter.com/CTCaer/status/1143103754964819968

[image]

CTCaer, 2019-06-30:
https://github.com/CTCaer/hekate/commit/bd7f572989de35d5ae7188ba967da56ca76fe7de

closed 2019-06-30 by CTCaer