CTCaer/hekate #228: Not working on latest system version

ThePsycrow, 2019-06-28:
When will we be able to use it as normal on 8.1.0?  Keeps having a problem with the pkg1 when trying to run the stuff I used to be able to run before 8.1.0.

fennectech, 2019-06-28:
Soon ™

fennectech, 2019-06-29:
The current way to use homebrew on 8.1.0 is by atmosphere’s fusee-primary until ctcaer finishes polishing hekate.

nicoboss, 2019-06-29:
They already have hekate working on 8.1.0 privately for 3 days - no idea why this changes still aren’t pushed yet. If the speculations are true and NYX is the successor of hekate then probably we get a working hekate like payload today: https://twitter.com/CTCaer/status/1144987055811944448
And if not CTCaer will probably have way more time in for hekate now that NYX seems to be done.
By the way this issue is a duplicate of https://github.com/CTCaer/hekate/issues/226 and so can be closed.

CTCaer, 2019-06-30:
https://github.com/CTCaer/hekate/commit/12f8f055eb94429ebd15395812d663b807fe2579

closed 2019-06-30 by CTCaer