CTCaer/hekate #231: Failure to backup BOOT0 during NAND backups/emuMMC creation with hekate/Nyx 0.8.0

suppai, 2019-06-30:
With the current Nyx release, every time I try to backup my NAND or create an emuMMC it fails to write to my SD card. For NAND backups/file-based emuMMC creation I get `fatal error when writing to the SD` when trying to backup BOOT0, and with partition-based emuMMC creation I get `Error writing 8192 blocks \@LBA` (it attempts to backup BOOT0 thrice and fails, then aborts). hekate 4.10.1 can create NAND backups without issue, and reading/writing works fine on PC, from fusee, in Horizon, etc. I tried multiple fresh formats of my SD with multiple different applications to make sure it wasn't at fault, and the issue persisted.

My SD card is a Samsung EVO+ 128GB microSDXC (MB-MC128D) formatted to FAT32, and my Switch is an unpatched console running firmware 1.0.0 (if that matters).

suppai, 2019-06-30:
Some additional info:

Perhaps this is due to SD card controller weirdness with certain cards *a la* the old Sandisk SD card issue, because the most recent iteration of this card, MB-MC128G *does* work, but MB-MC128D doesn't.

suppai, 2019-06-30:
Dumping fuses/bootrom/pk1/pk2/etc. works just fine-- just the NAND and emuMMC portions don't. Some screenshots: 

[image]

suppai, 2019-06-30:
Per m4xw's recommendation on Discord I reformatted again and made sure the cluster size was set to 32k; this did not make a difference and I encountered the same errors.

CTCaer, 2019-06-30:
Samsung should not have that problem. There's should be sth with when you enter the backup screen.
I'll investigate

suppai, 2019-06-30:
Thank you. Let me know if there's anything I can do to help! Fwiw, I purchased this card from Best Buy and also ran h2testw on it, so I'm fairly certain it is a legitimate Samsung card.

westonmyers, 2019-06-30:
I'm just curious, could you pull the cards' info via Nyx for both?

`Console Info > Storage & Battery info > microSD`

CTCaer, 2019-06-30:
Btw this fails instantly. The moment it tries to write. That's because the 0% text is not even initialized on the first runs.

suppai, 2019-06-30:
@westonmyers 

[image]

CTCaer, 2019-07-01:
So it's the older evo+ **U1**.
Haven't yet checked the whole code path to there, but I'll do it soon.

Geranospiza, 2019-07-01:
Same issue here. These are the sdcards I use. They both work in 4.10.1 and do their backups and stuff. Not working on 5.0.0 

[image]

CTCaer, 2019-07-01:
I'll need some of you to test this in discord, hopefully today.

ja3h, 2019-07-01:
I'm unsure if it's exactly related so I'm unsure it it'll even be of help (if not then I'm sorry), but I am also having issues with creating eMMC backups on the newest Hekate with a definitely-legit SanDisk Ultra 256gb mSD, of the common/newest type you can readily buy on Amazon (tested via FakeFlashTest to be doubly-sure). 
No other software or even hardware has faced any issues with read/write to these cards - not even OFW - except Hekate and I have tried two different ones of the same types of cards (formatted in both exFAT and FAT32, in all variations of cluster size) and it's a no-go, with the same issues each time though for some extra info if it helps any, FAT32 does SOMETIMES create BOOT0 and BOOT1 and other times it won't. On times it won't, if I retry a few times it eventually does. The other backups fail at random times even with verification disabled. 
The card-specific data on my cards are the exact same as suppai's (except the capacity and mine is App perf class A1), however I get a different error code to suppai's and that is error code (2).

suppai, 2019-07-02:
@CTCaer I am tart#4321 on Discord, feel free to DM me! (I'm in ReSwitched and Nintendo Homebrew if you need a mutual server, or you can post an invite here to one you're already in, or wherever)

CTCaer, 2019-07-02:
Can you join retronx, there's a build attached there.

This was probably fixed

suppai, 2019-07-02:
@CTCaer Joined; if the build you mention is hekate 5.0.1r3, the issue remains, though the error message is slightly different: `Error writing 8192 blocks @LBA 00000000` and `Fatal error (1) when writing to SD card`.

CTCaer, 2019-07-07:
Seems that the issue stems from UHS mode. Aka low voltage 1.8V path.
Forcing 3.3V makes them work.

CTCaer, 2019-12-08:
Please try again with v5.1.0.

I added a function that makes sure that PMIC changed the voltage and also APB clock (used in many things) is not stressed as before.

bellofiguoa, 2020-01-03:
Hi i had the same problem with 5.1.1 and after retrying a couple of times it managed to both backup the NAND and create the emuMMC. I have the same sd as suppay

CTCaer, 2020-01-05:
I found the issue to that problem and it will be fixed.

Anyone that wants to test that fix, ping me at RetroNX discord.

CTCaer, 2020-01-10:
The fix is confirmed to be working. Next version will have it

CTCaer, 2020-05-01:
Fixed with the new SDMMC driver

closed 2020-05-01 by CTCaer

migm16, 2020-09-23:
i get error code (5) when i try to make a boot0/1 backup in hekate, i have the raw gpp from a year ago and forgot to get the boot0/1 files untill today, i have upgraded once.

anroze00, 2022-11-22:
> Fixed with the new SDMMC driver

I man i got the same issue may get some help?

impeeza, 2022-11-22:
This error was corrected two years ago, if you have the "same" problem, try to update Hekate, and if the problem continue, please go to an appropriate support site, discord or GBATemp are good start.