CTCaer/hekate #232: Refuses to recognize Atmosphere logo bmp

NickHam13, 2019-06-30:
I am trying to set up a custom Atmosphère logo, however it won't show. I have another logo for stock and it works perfectly. My hekate-ipl.ini is here:

[Atmosphere]
fss0=atmosphere/fusee-secondary.bin
icon=bootloader/cfw.bmp
kip1patch=nosigchk
atmosphere=1

[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
icon=bootloader/stock.bmp

I tried moving the line for icon around, but it doesn't change. The difference is that my stock config displays correctly, however the Atmosphère icon won't display. Nonetheless, great work.

NickHam13, 2019-06-30:
I thought I'd pitch in here again to say that the background feature also doesn't work. 1280x720 photo named background.bmp in bootloader/res and it doesn't load in.

suppai, 2019-06-30:
Try opening your BMP in Windows MS Paint (**not** Paint 3D) and "Save As"-ing as a BMP again. In my experience, some BMPs (like those exported with GIMP and many online websites) are formatted incorrectly and won't work as splash screens.

NickHam13, 2019-06-30:
I tried that and unfortunately, as both 24-color bitmap and 256-color bitmap, it won't show up. Any other suggestions?

deurk, 2019-06-30:
Would love to know the specifics of the BMP files used too. Tried 32bit ABMP with Photoshop with no luck :(

CTCaer, 2019-06-30:
Only 32bit are supported.
And alpha is taken into account. So photoshop won't do it.

This can be used https://online-converting.com/image/convert2bmp/
with `32 bit (True color RGB)` or RGBA if you want the transparency

NickHam13, 2019-06-30:
Thanks, this fixed the problem.

closed 2019-06-30 by NickHam13

deurk, 2019-06-30:
Much better indeed.
[image]

NickHam13, 2019-07-01:
Oooh nice icons. You should try removing the text from the boot entry so that it's stated only in the icon.

deurk, 2019-07-01:
Will do, thanks for the tip.

Have a question though : transparency is handled by a grey background is present, any way to remove it @CTCaer ?

Thanks ;)

CTCaer, 2019-07-04:
@deurk The grey background is part of the window actually. The button has no body color.
Only borders that mask the selected image if the image is 192 x 192 or smaller

You can try finding the original resources for the icons used and resize them to 192 x 192.
No transparency. hekate will handle the corners. 
That will give you the aesthetically correct look you want.