CTCaer/hekate #233: "Failed to find applicable partition!"

imPRAGMA, 2019-06-30:
[image]

Just got 5.0.0 spinning up, I have a formatted factory reset nand, not even passed setup screen yet.
My SD card is fat32 formatted with guiformat.

NickHam13, 2019-06-30:
Create a FAT32 partition that's 31 GB. Use the rest of the SD for CFW. I had this same problem and this is what fixed it.

imPRAGMA, 2019-06-30:
Was this documented somewhere as instructions? 31gb sounds oddly specific no?

NickHam13, 2019-06-30:
To be precise, it has to be just big enough to house your NAND backup. This isn't documented anywhere, it's just something I tried that worked. I would go with 31 GB to be safe but you can make it exactly as big as your backup.

imPRAGMA, 2019-06-30:
If its just enough to hold nand, I guess it would be overall optimal to do 32gb, since that is the filesize of the nand storage size, if I had 21gb of that 32gb used, and made a backup, wouldnt I end up with a 21gb NAND backup, and if I create a partition of 21gb, id probably limit myself and result in errors.

suppai, 2019-06-30:
NAND backups are 29.1GB in size. As long as your partition is above that, it's fine. Either way, this is not a hekate/Nyx issue and should be closed.

NickHam13, 2019-06-30:
@suppai I agree, it should be closed now but how is this not relating to hekate and/or Nyx? He was getting an error in the payload and didn't know what was causing it, so naturally he assumed it was an issue.

imPRAGMA, 2019-06-30:
It isnt really a bug, just it should have been a tad bit documented, I know v5 just came out, but it would have been ideal.

closed 2019-06-30 by imPRAGMA

igoticecream, 2019-06-30:
i had this issue and solved it by incresing the emummc partition to 31.30 gb, if you use a Mac:

```
sudo diskutil eraseDisk FAT32 HOMEBREW MBRFormat /dev/disk4
sudo diskutil splitPartition /dev/disk4s1 FAT32 HOMEBREW 165.56g FAT32 EMUMMC 31.30g
```

(200gb sdcard)

CTCaer, 2019-06-30:
For now it expects a good sized primary partition (so unallocated space is a no also).
This will later change when support for resizing comes.

imPRAGMA, 2019-06-30:
[image]
This is my current SD card, yet it still gets the same error, am I missing something?

imPRAGMA, 2019-06-30:
Wait so I need the emummc partition to be fat32 formatted (which will result in there being 2 drives, 2 different drive letters)?

igoticecream, 2019-06-30:
Format both partitions as FAT32

imPRAGMA, 2019-06-30:
Perfect, working nicely. And I decided to go for 32gb because it would just be more even, and just look a tad bit better, I probably wont gain anything from it though.

CTCaer, 2019-06-30:
You don't need to format it. You just need to create a primary partition (raw - empty) in there. Unallocated is not yet supported.

pOOBAH73, 2019-11-23:
It only worked when i formatted both partitions as FAT32 here. Leaving the second (emuMMC) partition allocated (EXT4) but unformatted didn't work. Might be a Windows thing, depending on the partition tool you use.

Ericvf, 2019-11-27:
> It only worked when i formatted both partitions as FAT32 here. Leaving the second (emuMMC) partition allocated (EXT4) but unformatted didn't work. Might be a Windows thing, depending on the partition tool you use.

Thanks for sharing this. I can confirm this solved the same issue for me. The homebrew guide explicitly states that the partition should not be formatted but this seems wrong. Everything works now.

CTCaer, 2019-11-27:
It depends on the partition tool or if you are in Linux.

Basically some tools or Linux still apply partition type to unformatted partitions as x83 (Linux partition).

The trick is to format them as Linux swap, which is x82 and bypasses the check.

On the other hand it's a big no to format them as fat. Currently I don't wipe the first sectors and so the partition seems intact and this can corrupt emummc if you write anything in there.