CTCaer/hekate #234: Launch OFW from exFAT card

UdoChudo, 2019-06-30:
So, when i try to load stock OFW from my sd card in exFAT fs, i got Nin logo and black screen, When i try boot to Atmo it booting and working good. Put another SD Card in FAT32 format and this boot to OFW without problem. 
Question, how to fix booting to OFW through exFAT sdcard ? 
Hekate 5.0
Nyx 0.8.0
Atmo 0.9.2

CTCaer, 2019-06-30:
You have emuMMC enabled?

If yes, you need to add `emummc_force_disable=1` to your Stock profile. There are some incompatibilities with stock emuMMC, which I yet to investigate fully.

UdoChudo, 2019-06-30:
Oh..ok, thx
I need add this
> emummc_force_disable=1

for all OFW or only for latest ?

CTCaer, 2019-06-30:
For all fw versions when running stock, if you have emuMMC enabled.

UdoChudo, 2019-06-30:
> For all fw versions when running stock, if you have emuMMC enabled.

Oh, ok. Thx. Pls, can read second issue :)

closed 2019-06-30 by UdoChudo