CTCaer/hekate #235: SD card partitions become unreadable in Linux

westonmyers, 2019-06-30:
Environment
Distro: Arch
Kernel: 5.1.15
util-linux: 2.34-2
Desktop: KDE Plasma 5.16.2-1

SD card at `/dev/sdc`; previously showing `/dev/sdc1` and `/dev/sdc2` under `lsblk`, etc.
Now though:

`lsblk`
```
NAME MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sdc    8:32   1 366.8G  0 disk 
```

`sfdisk -l /dev/sdc`
```
Disk /dev/sdc: 366.84 GiB, 393870311424 bytes, 769277952 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5003b316

Device     Boot     Start       End   Sectors   Size Id Type
/dev/sdc1            2048 705230847 705228800 336.3G  b W95 FAT32
/dev/sdc2       705230848 769271807  64040960  30.6G ee GPT

```

I was able to use `sfdisk` to easily edit the partition type of `/dev/sdc2` from `ee` to something more mbr-like, such as type `ef`, and get back up and running.

```
sfdisk -d /dev/sdc > table
vim table
sfdisk /dev/sdc < table
```

`lsblk`
```
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sdc      8:32   1 366.8G  0 disk 
├─sdc1   8:33   1 336.3G  0 part 
└─sdc2   8:34   1  30.6G  0 part 
```

`sfdisk -l /dev/sdc`
```
Disk /dev/sdc: 366.84 GiB, 393870311424 bytes, 769277952 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5003b316

Device     Boot     Start       End   Sectors   Size Id Type
/dev/sdc1            2048 705230847 705228800 336.3G  b W95 FAT32
/dev/sdc2       705230848 769271807  64040960  30.6G ef EFI (FAT-12/16/32)
```

The three lines that probably need to be changed.

https://github.com/CTCaer/hekate/blob/e6d1a310452d29b20de0a9b3cb4e0654ca9f1e3f/nyx/nyx_gui/frontend/fe_emummc_tools.c#L565
https://github.com/CTCaer/hekate/blob/e6d1a310452d29b20de0a9b3cb4e0654ca9f1e3f/nyx/nyx_gui/frontend/fe_emmc_tools.c#L74
https://github.com/CTCaer/hekate/blob/e6d1a310452d29b20de0a9b3cb4e0654ca9f1e3f/nyx/nyx_gui/frontend/gui_emummc_tools.c#L279

ckurtz22, 2019-06-30:
Yeah, CTCaer is aware of this. The fix is like you said, change those 0xEEs into 0xEF, recompile and upload to SD card, and then remigrate emuMMC

CTCaer, 2019-06-30:
Yep, this will be fixed.

null-device, 2019-06-30:
Here's a recompiled nyx.bin with the changes above:

[nyx.bin.zip](https://github.com/CTCaer/hekate/files/3342924/nyx.bin.zip)

chillyshackk, 2019-07-01:
Thank you for this, will try this now, I was wondering what was going on with the partitions and ended up retrying on 3 different sd cards..!

chillyshackk, 2019-07-01:
Worked perfectly thanks! Uploaded the bin via ftpd, booted in hekate/nyx, pressed migrate emuMMC, tested that it still worked then I put the card back on my pc and the partitions were there!

CTCaer, 2019-07-01:
FYI: This also breaks Lakka and L4T ubuntu.
Basically whatever needs the "boot" partition and it's Linux based.

JakeTripplJ, 2019-07-02:
Another way to make the partitions visible in Linux is to use parted's rescue function. For example, all I had to do was 
`parted /dev/mmcblk0` (or whatever your SD card is recognized as)
and then 
`rescue` 
at the next prompt. When it asks for start and end points, I just had it scan the entire card. It found the FAT32 partition almost instantly and I was able to use it normally after that. The rest of the SD card shows up as unallocated, but I labeled it as a RAW partition just so I don't accidentally overwrite anything.

westonmyers, 2019-07-02:
Rescuing requires more input, takes a ton of time (Comparatively), and is completely unnecessary. You _can_ do it, but there's no reason to do so in this context. The partition table is "fine." It's not as though the partition table is corrupt or what-have-you. Offsets are correct, etc. The partition type just needs to be swung.

CTCaer, 2019-07-06:
Partitions are now marked as **E0**
Running migrate will fix them.

closed 2019-07-06 by CTCaer